open-ephys/plugin-GUI #327: Adding custom plugin

holmescharles, 2019-10-30:
A labmate wrote a custom plugin for my setup a long time ago. I just updated open-ephys, but the build structure is much different. I used to be able to just copy the ".so" file of the custom plugin into the plugins directory, but that does not seem to work.

If I have a built ".so" file, where can I put this so open-ephys "loads" it? (I am working on Ubuntu if that helps). Thanks!

aacuevas, 2019-10-30:
The location for plugins is still the same, as the way to load them. However, in last version there were some changes to the plugin API that requires plugins to be built again against it. We try to avoid making such changes as much as possible, but sometimes they are unfortunately required. It is probable your plugin was built for the previous API version and needs to be rebuilt.

Looking at the console when starting the GUI should confirm you if that is indeed the case or nor, as the plugin should be listed but showing an error message.

holmescharles, 2019-11-04:
It didn't show, but I have downgraded to an older version of the plugin GUI for now.

closed 2019-11-04 by holmescharles