open-ephys/plugin-GUI #328: cmake install script error

catubc, 2019-11-21:
Hi.

I was not able to install the GUI with the default CMakeLists.txt file.  This is the error I got;

```
(open_ephys) cat@cat-ThinkPad-T520:~/code/plugin-GUI/Build$ cmake -G "Unix Makefiles" OE_DONT_CHECK_BUILD_PATH=True -DCMAKE_BUILD_TYPE=Release /home/cat/code/plugin-GUI/CMakeLists.txt 
CMake Error at CMakeLists.txt:18 (message):
  Please run cmake inside the Build folder to build the binaries in the
  predefined paths.

  Set the OE_DONT_CHECK_BUILD_PATH cmake variable to TRUE to disable this
  error

-- Configuring incomplete, errors occurred!
```

However, when I commented out the conditional (see below), I was able to build and run the GUI:

```
#if ((NOT ${CMAKE_CURRENT_BINARY_DIR} STREQUAL ${CMAKE_CURRENT_SOURCE_DIR}/Build ) AND (NOT OE_DONT_CHECK_BUILD_PATH))
#message(FATAL_ERROR "Please run cmake inside the Build folder to build the binaries in the predefined paths.\nSet the OE_DONT_CHECK_BUILD_PATH cmake variable to TRUE to disable this error")
#endif()
```

I hope I will not have other issues later on.
Thanks!

anjaldoshi, 2019-11-21:
Hi, 

A better way to generate Build files would be: `cmake -G "Unix Makefiles" DCMAKE_BUILD_TYPE=Release ..`

This is because you are already inside the Build folder, so you don't have to set the `OE_DONT_CHECK_BUILD_PATH` to true. Also, you don't need to specify the full path for CMakeLists.txt. You can just type `..` to go one level up the Build folder, where the CMakeLists.txt file is. Cmake will find that file on its own and use it.

Also, no, this will not cause an issue in you case because you generated the Build files at the correct location. So, installing new plugins should will not be a problem (plugins require the GUI to be installed at a predefined path).

aacuevas, 2019-11-21:
Also, any cmake variable is set with the -D flag. so If you needed to use OE_DONT_CHECK_BUILD_PATH you need to specify it with `-DOE_DONT_CHECK_BUILD_PATH=True`

closed 2019-11-21 by catubc

catubc, 2019-11-22:
Thanks.  But the default

` cmake -G "Unix Makefiles" DCMAKE_BUILD_TYPE=Release ..`

didn't work on my setup.  Not sure what I had done wrong. 

I will assume that this is solved for now.

Thanks for the input.

anjaldoshi, 2019-11-22:
Oh, my bad. I had a typo in my original comment. Just like Aaron said, cmake variables need to be set with the `-D` flag, and my original comment is missing a hyphen(-) preceding the D for setting CMAKE_BUILD_TYPE. So, the correct command is:

`cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release ..`

I hope this helps.