open-ephys/plugin-GUI #329: cannot bind packed field error when building from source in Linux

teristam, 2019-11-25:
Hi,

I am trying to build from source in Ubuntu. I should have all the dependencies installed, but when I execute `make`, i receive the following error:

cannot bind packed field ‘((juce::PixelARGB*)this)->juce::PixelARGB::<anonymous>.juce::PixelARGB::<unnamed union>::comps[0]’ to ‘juce::uint8&’ {aka ‘unsigned char&’}

It is somehow related to this line:
```
 #if JUCE_GCC
    // NB these are here as a workaround because GCC refuses to bind to packed values.
    forcedinline uint8& getAlpha() noexcept           { return comps [indexA]; }
    forcedinline uint8& getRed() noexcept             { return comps [indexR]; }
    forcedinline uint8& getGreen() noexcept           { return comps [indexG]; }
    forcedinline uint8& getBlue() noexcept            { return comps [indexB]; }
   #else
    forcedinline uint8& getAlpha() noexcept           { return components.a; }
    forcedinline uint8& getRed() noexcept             { return components.r; }
    forcedinline uint8& getGreen() noexcept           { return components.g; }
    forcedinline uint8& getBlue() noexcept            { return components.b; }
   #endif; } 
```
is there any variable I need to set before building? Thanks!

anjaldoshi, 2019-11-25:
Hi,

Thanks for providing this detailed information. The error shows that it is a JUCE library error. Just a quick search led me to an issue in the JUCE library repo, which says that it has to do something with your GCC compiler version. Our current version of JUCE doesn't support GCC 9. Can you check and let me know what version of GCC compiler you have by typing `gcc --version` in your terminal?

teristam, 2019-11-25:
Hi, my version is gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008. Guess I need to change my gcc version then, may I ask which version should I use?

anjaldoshi, 2019-11-25:
Can you try  running your make as follows:

`make CC=gcc-9 CXX=g++-9`

and let me know if you still face that error?

teristam, 2019-11-25:
Hi,

Unfortunately `make CC=gcc-9 CXX=g++-9` doesn't work for me. But I find that setting
```
export CC=gcc-8
export CXX=g++-8
```
before calling `cmake` works for me.

Thank you so much!

anjaldoshi, 2019-11-25:
Oh, great! You're welcome! We appreciate your input on explicitly setting compiler version before calling cmake.

GCC 9 compatibility issues has been fixed in the latest version of JUCE. Updating the JUCE library, provided with the GUI, to the latest version is already a part of our TO-DO list.

closed 2019-11-26 by teristam