open-ephys/plugin-GUI #334: Issues building from source

angelonc, 2020-01-23:
Hi, I'm trying to upgrade to the latest version of the source code, and was having several issues, first, in making the Build files, the recommended command: 

`cd Build
cmake -g "Visual Studio 12 2013 Win64" ..`

fails, saying that I need to run it from the Build directory, even though I already am in that directory. I was able to get it to work by setting the option -DOE_DONT_CHECK_BUILD_PATH=True after the command above.

This made the build files, which I was able to successfully build in VS2013, but when I run the application, I do not have the Rhythm FPGA Module.

Any ideas?

angelonc, 2020-01-23:
Ok, since I had previously built an older version of the GUI in the folder I was working in, I cloned a fresh fork of the repo and started over. In the fresh clone, the cmake commands work fine, but I still can't see the Rhythm FPGA Module....

angelonc, 2020-01-23:
Ok, I fixed it, but only by adding the okFrontPanel.dll file from a previously working release to my Build\Release folder. The okFrontPanel.dll in the Resources\DLL folder does not work either... maybe there is something wrong with the .dll file in the current github version? Or a compatibility issue?

I'm running on
Windows 10 x64

medengineer, 2020-01-28:
Hi,
I'm unable to reproduce this case on my end; I'm also on Win 10 x64 + OpenEphys board. The okFrontPanel.dll hasn't been updated in a couple years, is the working version older than that? If you're able to run open-ephys from the command line, could you copy over the output when dropping the working Rhythm FPGA module into the signal chain. For example, mine shows: 
---- Intan Technologies ---- Rhythm RHD2000 Controller v1.41 ----
FrontPanel DLL loaded.  Built: Nov 11 2016  13:08:03

angelonc, 2020-01-28:
I'm not sure how to run OEPs from the command line, are there instructions somewhere?

medengineer, 2020-01-28:
If you have GitBash or ConEmu you can navigate to Build/Release directory and run open-ephys using the command './open-ephys'

aacuevas, 2020-01-28:
Alternatively, you can create a shortcut to the executable with an option that will open a console. You can find the instructions at the bottom of https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/491625/User+documentation

angelonc, 2020-01-31:
When I run it from command line with the .dll file from Resources/DLLs/Win64 I get the following errors:

`Loading Plugin: RhythmNode...  DLL Load FAILED
Loading Plugin: SerialInput... Loaded with 1 plugins
Loading Plugin: SpikeSorter... Loaded with 1 plugins

Loading window bounds.

C:\Users\St. Anger\Documents\GitHub\plugin-GUI\Source\Processors\PluginManager\PluginManager.cpp:181: Failed to load plugin DLL: DLL Error 0x7e
C:\Users\St. Anger\Documents\GitHub\plugin-GUI\Source\Processors\PluginManager\PluginManager.cpp:181: Failed to load plugin DLL: DLL Error 0x7e
`

medengineer, 2020-01-31:
What's the output when you use the 'working' okFrontPanel.dll?

angelonc, 2020-01-31:
> $ ./open-ephys
> 
> 
> Created processor graph.
> 
> Got audio device.
> 
> Audio device name: LG ULTRAWIDE (Intel(R) Display Audio)
> Audio device type: Windows Audio
> Audio output channels: 2
> Audio device sample rate: 44100
> Audio device buffer size: 1024
> 
> Created audio component.
> Created message center.
> Created info label.
> Created graph viewer.
> Adding tab with index 1
> Adding tab with index 2
> Created data viewport.
> Created filter viewport.
> Created control panel.
> Created filter list.
> Created plugin manager
> Loading Plugin: ArduinoOutput... Loaded with 1 plugins
> Loading Plugin: BasicSpikeDisplay... Loaded with 2 plugins
> Loading Plugin: CAR... Loaded with 1 plugins
> Loading Plugin: ChannelMappingNode... Loaded with 1 plugins
> Loading Plugin: EvntTrigAvg... Loaded with 1 plugins
> Loading Plugin: FilterNode... Loaded with 1 plugins
> Loading Plugin: IntanRecordingController... Loaded with 1 plugins
> Loading Plugin: LfpDisplayNode... Loaded with 1 plugins
> Loading Plugin: LfpDisplayNodeBeta... Loaded with 1 plugins
> Loading Plugin: PhaseDetector... Loaded with 1 plugins
> Loading Plugin: PulsePalOutput... Loaded with 1 plugins
> Loading Plugin: RecordControl... Loaded with 1 plugins
> Loading Plugin: Rectifier... Loaded with 1 plugins
> Loading Plugin: RhythmNode... Loaded with 1 plugins
> Loading Plugin: SerialInput... Loaded with 1 plugins
> Loading Plugin: SpikeSorter... Loaded with 1 plugins
> 
> Loading window bounds.
> 
> Loading processor graph.
> Clearing signal chain.
> Item dropped at insertion point 0
> Creating from plugin info...
> Rhythm FPGA(1)::Rhythm FPGA
> C:\Users\St. Anger\Documents\GitHub\plugin-GUI\Build\Release
> ---- Intan Technologies ---- Rhythm RHD2000 Controller v1.41 ----
> 
> 
> FrontPanel DLL loaded.  Built: Mar 17 2014  11:47:26
>

closed 2020-12-16 by jsiegle

KatharineShapcott, 2021-09-30:
Hi all, I ran into the exact same problem. For me there's nothing wrong with the created okFrontPanel.dll file, it just needs to be moved into the Release folder. 

I suspect a problem linking the directories in cmake. I'm using 3.18, what version is usually being used?