open-ephys/plugin-GUI #338: Crash when changing prepended text with Binary Format

markschatza, 2020-02-05:
Hi,

We recently tried to swap from continuous to binary file format. We are currently using network events to break up segments of our recordings into chunks (closed-loop, ERPs, Raw). Using network events we send commands to OE using the following logic.

```
stopRecording()
prependText("blah")
startRecording()
```

With continuous file format this works fine, but it seems to be causing OE to crash with binary. I haven't looked into it yet and will update as I do.

markschatza, 2020-02-06:
Hmm, doing this in a very basic environment doesn't cause a crash. Once I find out what I'm doing different in our experiments and how to reliably replicate the issue I will post another update.

open-ephys, 2020-02-06:
Ok, thanks for looking into this! We will test this out next week if you haven't solved it by then.

markschatza, 2020-03-04:
Finally got around to looking at this again. Doesn't crash when using file reader(which is what I used to try to replicate), but does while using Rhythm FPGA. Will be testing more.

markschatza, 2020-03-05:
I believe the machine we are trying to run is too slow at stopping the recording thread, as I'm sure there is more overhead in stopping binary vs continuous format.  This should be fine (albeit slow) but when trying to ask if the user wants to continue waiting it crashes at line 359 in RecordNode.cpp trying to run 

``` c++
if (AlertWindow::showOkCancelBox(AlertWindow::WarningIcon, "Record Thread timeout",
					"The recording thread is taking too long to close.\nThis could mean there is still data waiting to be written in the buffer, but it normally "
					"shouldn't take this long.\nYou can either wait a bit more or forcefully close the thread. Note that data might be lost or corrupted"
					"if forcibly closing the thread.", "Stop the thread", "Wait a bit more"))
				{

```

For some reason this showOkCancelBox() is trying to access some sort of locked message class? Might be a multi-threading juce bug?

Anyways, upping the timeout from 2000 to 4000 in the line above was a large enough timeout for it finish its job.

jsiegle, 2020-03-06:
What are the specs of the computer you're using? 

@aacuevas, is it reasonable to increase the timeout period for this?

aacuevas, 2020-03-06:
The timeout value is somewhat arbitrary, so it can be increased. I set it to 2 seconds because it took for my machine around that time to properly close the thread for a couple thousand channels. But the dialog is there for cases in which that is not enough. A crash should never happen.

I have tested decreasing the timeout to an absurdly low value so it always triggers timeout and the dialog still works for me, so it is weird that it is crashing for you.

Does it fail when manually stopping recording or only when using networkevents as stated in the report? Are you sending the three commands in fast sequency or there is time between them?

markschatza, 2020-03-09:
Looks like the crash is only when attempting to open the AlertWindow when the stop recording is sent from networkevents. The three commands being fast or delayed have no effect. Something with AlertWindow and how networkevents is handling stop recording are not agreeing.

The dialogue works fine when using the manual stop recording for me as well.

jsiegle, 2022-07-30:
This should be fixed in the latest version of the GUI, but if not, please re-open this issue.

closed 2022-07-30 by jsiegle