open-ephys/plugin-GUI #339: Channel name not shown correctly when using flat binary format in FileReader

teristam, 2020-02-13:
Hi,

I am converting some of our old recordings in the Open Ephys format to the flat binary format so that we can replay them in the Open Ephys GUI using FileReader. I managed to create the folder structure and the `oebin` file. The data are able to load correctly but I encounter some  problems:

1. The channel name in LFP Viewer always display 'CH1', 'CH2' etc no matter how I set the `channel_name` attribute in the oebin file. We have some ADC channels so we would like to clearly distinguish between them.
2. The LFP Viewer doesn't seem to distinguish between normal data channel and ADC data. Is there a way to specify a ADC or AUX channel in the flat binary format?

Any help will be much appreciated! Thanks a lot!

my oebin file attached as txt
[structure.txt](https://github.com/open-ephys/plugin-GUI/files/4200182/structure.txt)

jsiegle, 2020-02-17:
1. Unfortunately, specifying different channel types isn't possible with the current File Reader. We are planning some major upgrades to the File Reader over the next few months, and this will be one of the new features we add.

2. This seems like a bug. @anjaldoshi, can you take a look at this?

teristam, 2020-02-17:
Thank you for the reply! I managed to compile the GUI from source and have looked at the source code of FileReader. But I don't seem to figure out where is the data channel got created.  I changed the `getDefaultNumDataOutputs` in `FileReader.cpp` as follows:

```
int FileReader::getDefaultNumDataOutputs(DataChannel::DataChannelTypes type, int subproc) const
{

    switch(type){
        case DataChannel::HEADSTAGE_CHANNEL:
            return 16;
        case DataChannel::ADC_CHANNEL:
            return 8; //hard-coded for now 
        case DataChannel::AUX_CHANNEL:
            return 3;
        default:
            return 0;
    }
}```

And I seem to be able to create the ADC channels. But how can I specific the order of the ADC and AUX channel in reference to the channel index in the data buffer (i.e. which channel is ADC and AUX)? Thanks in advance!

teristam, 2020-02-17:
Just to report back that i managed to have a quick solution working at my [fork](https://github.com/teristam/plugin-GUI/commit/eedf726b53f49587123cefbb128fcba423b3f402). It is a quick and dirty solution to detect channel type from the name, but if you are interested, I am happy to make a pull request. Thank you!

anjaldoshi, 2020-02-17:
Hey @teristam, I'm not able to repro the bug you mentioned in 2. I loaded in a sample oebin, started acquisition, stopped it, clicked on "F:" and reloaded the same file, and it worked perfectly fine. The timestamp got reset back to 00:00, and then when I started acquisition, it started reading the file from the beginning. 

Can you please provide me with proper details on how to reproduce this bug?

teristam, 2020-02-17:
@anjaldoshi , I think it doesn't reload the information from the oebin file. After you stop acquisition, and change one of the channel name in the oebin file manually, and reload the oebin file via pressing F:, the channel name doesn't change. I guess for most users it won't cause any problem, but for advanced user where they need to play around with manually converting other filetype into flat binary format, it may lead to some confusion.

anjaldoshi, 2020-02-17:
Thanks for the details @teristam. I was able to repro the bug. It is definitely a bug that needs to be addressed. I think it could be either an LFP Viewer or a File Reader issue. The channel names being displayed on the screen are fetched and updated by the LFP Viewer, but it gets reported by the processor before it in the signal chain (file reader in this case). So, I will have to take a look into both of  them to sort this issue out.