open-ephys/plugin-GUI #340: Missing File Reader code in source

samjlevy, 2020-02-26:
Hi all, 
I'm a new user to this environment, so please forgive me if I'm missing something obvious. It seems like the source code available on GitHub, used to build the executable in CMake, doesn't include all the necessary files for the File Reader module to run the way it does in the pre-made executable. The FileReader module doesn't work for me when built from source, and the files for the pre-made executable, I see dlls that reference the KWIK format, among others, which do not appear when I build from source. Maybe this would be fixable by copying over those dlls after building, but i doesn't, and I can't tell in the source code for the file reader where it gets its list of supported file types.

medengineer, 2020-02-26:
Hi,
The pre-made executable is a combination of the plugin-GUI source code and the HDF5 plugins source code from here: https://github.com/open-ephys-plugins/HDF5Plugins. In order to re-create the pre-made executable, you will need to build the plugin-GUI from source, then build the HDF5Plugins which will generate and install the KWIKFormat.dll.

anjaldoshi, 2020-02-26:
Also, yes, you can just copy the KwikFormat and hdf5 dlls from the pre-made executable's plugins and shared directories to those of the executable built from source, and it will work as intended.

samjlevy, 2020-02-26:
> Also, yes, you can just copy the KwikFormat and hdf5 dlls from the pre-made executable's plugins and shared directories to those of the executable built from source, and it will work as intended.

Can you offer a little more detail? The dlls in plugins from the pre-made executable seem to only be the KWIK and NWB options. There are a few hdf5 dlls in the main folder, though I tried copying those in after build and still nothing; the dialog box for selecting files is also fixed on oebin files. Are there files that maybe need to move before the build is created?

samjlevy, 2020-02-26:
Also tried building the HDF5 plugins from source and I'm getting an error during build from the install project in Visual Studio that some files are missing in the HDF5 plugin source folders.

Does there exist a step-by-step, beginner oriented guide to building projects from source in Visual studio? I suspect some of my challenge is coming from having never done anything like this before, probably there are some basics I'm missing trying to interpret the open ephys wiki

anjaldoshi, 2020-02-26:
You just need to copy the **KWIKFromat.dll** from the plugins folder of the pre-built executable to the plugins folder of your source-built executable for the _.kwd_ option to show up in the file explorer dialog box. The plugins folder of your source-built executable is going to be under Build/Release (or Debug depending on which config you built it in). Something like this: `plugin-GUI/Build/Release/plugins`. Once you do that, run the executable and the File Reader will allow you to select a _.kwd_ file as an input. Follow the same procedure for NWBFormat.dll if you want to read NWB files using the File Reader.

anjaldoshi, 2020-02-26:
Regarding your HDF5 plugin error, can you post the error log here? You can just copy paste the Build output from Visual Studio. 
Also which version of Visual Studio are you using?

samjlevy, 2020-02-27:
Weird, I've been trying exactly that procedure of copying over the pre-built KWIFformat.dll and my file reader module still won't let me switch the file type extension it's looking for. The only way to get it to recognize KWIK files is to change the line in the C++ source where it references oebin, but even then if I copy over the dll it still won't load the file. 

However this shakes out, I appreciate your taking the time to help me out. 

I'm using Visual studio 2019. Just in case it matters, my directories are set up so that the plugin-GUI code is in ... Open Ephys\plugin-GUI-master\plugin-GUI-master, and the HDF5 code is in ... Open Ephys\plugin-GUI-master\plugin-GUI-master. The error log when I build from the installer is as follows:

1>------ Build started: Project: OpenEphysHDF5Lib, Configuration: Release x64 ------
1>HDF5FileFormat.cpp
1>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\libs\include\H5Exception.h(89,29): warning C4251: 'H5::Exception::detail_message': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'H5::Exception'
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\include\xstring(4448): message : see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
1>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\libs\include\H5Exception.h(90,24): warning C4251: 'H5::Exception::func_name': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'H5::Exception'
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\include\xstring(4448): message : see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
1>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\OpenEphysHDF5Lib\Source\HDF5FileFormat.h(27,10): fatal error C1083: Cannot open include file: 'CommonLibHeader.h': No such file or directory
1>Done building project "OpenEphysHDF5Lib.vcxproj" -- FAILED.
2>------ Build started: Project: KWIKFormat, Configuration: Release x64 ------
3>------ Build started: Project: NWBFormat, Configuration: Release x64 ------
2>KwikFileSource.cpp
2>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\libs\include\H5Exception.h(89,29): warning C4251: 'H5::Exception::detail_message': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'H5::Exception'
2>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\include\xstring(4448): message : see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
2>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\libs\include\H5Exception.h(90,24): warning C4251: 'H5::Exception::func_name': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'H5::Exception'
2>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\include\xstring(4448): message : see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
2>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\KWIKFormat\Source\FileSource\KwikFileSource.h(27,10): fatal error C1083: Cannot open include file: 'FileSourceHeaders.h': No such file or directory
2>OpenEphysLib.cpp
2>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\KWIKFormat\Source\OpenEphysLib.cpp(24,10): fatal error C1083: Cannot open include file: 'PluginInfo.h': No such file or directory
2>HDF5Recording.cpp
3>NWBFormat.cpp
2>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\KWIKFormat\Source\RecordEngine\HDF5Recording.h(27,10): fatal error C1083: Cannot open include file: 'RecordingLib.h': No such file or directory
2>KWIKFormat.cpp
3>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\NWBFormat\..\OpenEphysHDF5Lib\Source\HDF5FileFormat.h(27,10): fatal error C1083: Cannot open include file: 'CommonLibHeader.h': No such file or directory
3>NWBRecording.cpp
2>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\KWIKFormat\..\OpenEphysHDF5Lib\Source\HDF5FileFormat.h(27,10): fatal error C1083: Cannot open include file: 'CommonLibHeader.h': No such file or directory
2>Generating Code...
2>Done building project "KWIKFormat.vcxproj" -- FAILED.
3>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\NWBFormat\Source\NWBRecording.h(27,11): fatal error C1083: Cannot open include file: 'RecordingLib.h': No such file or directory
3>OpenEphysLib.cpp
3>C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\HDF5Plugins-master\NWBFormat\Source\OpenEphysLib.cpp(24,10): fatal error C1083: Cannot open include file: 'PluginInfo.h': No such file or directory
3>Generating Code...
3>Done building project "NWBFormat.vcxproj" -- FAILED.
========== Build: 0 succeeded, 3 failed, 2 up-to-date, 0 skipped ==========

anjaldoshi, 2020-02-27:
- That is weird. To check whether the KWIKFormat.dll loaded properly or not, try running the executable via PowerShell (or any other terminal). Go to the directory where your executable is, and press Shift + right click.  Select "Open PowerShell window here" from the context menu and run this command  `.\open-ephys.exe --console`
This will run the executable along with a new console window. Look at the output of the console, check whether the KWIKFormat.dll loaded properly or not. If it loaded properly, there should be a print statement like this: `Loading Plugin: KWIKFormat... Loaded with 2 plugins`. If it didn't load properly, then it will print an error code. Copy and paste that error code here.

- Regarding your Build errors for HDF5 plugins, your're facing these errors because the HDF5plugins  needs the JUCE headers and GUI headers to build the source. The path you've cloned HDF5plugins at is incorrect and thus it is not able to refer those headers. If the GUI source is located in /usr/src/open-ephys/plugin-GUI the plugin should be in /usr/src/open-ephys/PuginName. If the plugin source is located in a different location, you can generate the build files by defining the
-DGUI_BASE_DIR path while running the cmake command. So, in your case, the cmake command is going to look something like this: 
` cmake -G "Visual Studio 16 2019" -A x64 -DGUI_BASE_DIR=C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master ..`
Try running the above mentioned command from your HDF5Plugins' Build directory assuming that you GUI source is under "C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master". Also, before running the cmake command, make sure that you delete "CMakeCache.txt" and "CMakeFiles" directory from the HDF5Plugins' Build directory.

Let me know if you have any other questions or concerns. I'm always happy to help.

samjlevy, 2020-02-28:
When I copy the KWIK dll into the plugins folder in the plugin-GUI release and run the executable from console as you suggested, I got the error that you predicted: 

Loading Plugin: KWIKFormat... C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\plugin-GUI-master\Source\Processors\PluginManager\PluginManager.cpp:181: Failed to load plugin DLL: DLL Error 0x7e
 DLL Load FAILED

Sorry for being slow about this: I have the directories setup with: 
C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master
and:
C:\Users\Sam\Documents\Open Ephys\HDF5plugins

The CMake powershell command worked fine and I opened the solution in Visual Studio 2019, and then while building I'm getting an error that says:
LNK1112: module machine type 'x86' conflicts with target machine type 'x64'
I've tried a bunch of things to get past this, like changing the build type to DLL so that I can change the target machine in the Linker settings to x86; building the original GUI to x64 and x86; changing the CMake powershell command to x86, but that one then fails.

Next steps? Continuing thanks for your efforts

anjaldoshi, 2020-02-28:
Yeah, no problem! 
Can you make sure that you also copied over the `OpenEphysHDF5Lib.dll` to the `shared` folder which is at the same level as the `plugins` folder where you copied the `KWIKFormat.dll`. If not, then doing this might fix you DLL load error as KWIKFormat is dependent on OpenEphysHDF5lib.

Regarding your build error, make sure the build config for both, GUI and HDF5plugins, is set to Release and x64 (see image below)
[image]

Also, in the DLL load error, I see that your GUI source is at 
`C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master\plugin-GUI-master`
However, later you say that your GUI source is at 
`C:\Users\Sam\Documents\Open Ephys\plugin-GUI-master`

So, make sure you specify the correct path in `-DGUI_BASE_DIR` for the HDF5Plugins cmake command.

Take your time and keep me posted.

samjlevy, 2020-03-09:
Update on my end: 
Regarding the directory change, I moved code into folders that had fewer "redundant" levels in the directory tree, usually that emerged from decompressing a zip folder. 

It looks like a lot of my trouble was making sure that everything was building using x64, both in CMake and in Visual studio. Once I did that, I was able to build the KWIK file reader and it works as expected. 
I then ran into some trouble getting the FFTW plug-ins to install, where even after following the same steps to get the HDF5 plugins to work, I got the dlls to build but they didn't move automatically to whatever folder they're supposed to end up in and a plug in dependent on those couldn't find them. 

My solution to all of these issues is that the lab which makes a plug in that I was trying to use all along also has a .bat file which will build Open Ephys plugin GUI along with their plugins and all the necessary support plugins , and this includes the KWIKformat and HDF5 plugins. They were very kind and updated this bat to work with Visual Studio 2017 and 2019. 
https://github.com/tne-lab/oep-installation
For now, this has solved all the issues I was facing and gotten me to a point where I can move forward with my project.

anjaldoshi, 2020-03-09:
It is great to hear that you were able to sort your issues out. I understand that for people who build  Open Ephys GUI and its plugins from source for the first time, it can be a bit overwhelming. We are working on completely redoing the documentation for the GUI with detailed information on how to get involved with the development of the GUI and its plugins. Hopefully, that will help new developers solve the issues related to building the GUI and plugins, as well as other development issues one might face. 

We are also working on a "plugin installer" for the GUI that will help people who don't want to build the plugins from source by providing a neat interface to "one-click" download and install plugins (and their dependencies) via a centralized server. This will help in reducing the amount of time spent to download and install plugins.

I hope I've been helpful. If I can be of any further assistance, please, let me know.

closed 2020-03-12 by samjlevy