open-ephys/plugin-GUI #342: About best practise to develop and debug an open-ephys plugin

teristam, 2020-03-06:
Hi,

I am developing a plugin for Open-Ephys and I am working in Visual Studio. May I ask for any recommandation for the most efficient way to debug a plugin? Now my workflow looks something like this

1. Edit the plugin source code in Visual Studio
2. Build the ALL_BUILD project to compile and install the plugin DLL into the open ephys plugin folder
3. Open open-ephys, load a default config and try the plugin

While it kind of works, but I run into some severe limitations

1. I cannot debug the plugin code directly using a debugger, because open-ephys and the plugin are in a different VS solution
2. I have to manually open the configuration file everytime I run open ephys. Is there a way to call open-ephys from commandline and specify that it always open a certain configuration file?

Thank you so much!

medengineer, 2020-03-06:
Hi,

When building your plugin, you can use the INSTALL project to compile and install your plugin's .dll file into the plugins folder of the plugin-GUI. In order to debug the plugin, you will need to run the plugin code in either Debug or RelWithDebInfo configuration mode to be able to step through your plugin's code. RelWithDebInfo mode runs faster than Debug, with the tradeoff of that some variables will not be available to examine while debugging. While open-ephys.exe is running, you debug the plugin code by selecting Debug -> Attach To Process in Visual Studio and selecting open-ephys.exe to attach to. 

If you are running the same configuration every time to test your plugin, it's best to just enable the ReloadOnStartup option from the File menu of the plugin-GUI. This will load in the lastConfig.xml file from your previous session. Otherwise, you can call open-ephys.exe <your_config>.xml if you have different configurations you'd like to test.

teristam, 2020-03-07:
Thank you so much! Those steps help a lot!

closed 2020-03-07 by teristam

markschatza, 2020-03-08:
Hey,

A quick tip! If you "Open a Project or Solution" in Visual Studio you can open your open-ephys.exe file and only need to click start to run it. I find it to be much less clicking than the "Debug -> Attach to Process" method!

njantony1, 2022-02-04:
Hello! Sorry for posting to an old thread... For some reason, when I try to do this, I get an error message that says "Unable to attach to the process. A debugger is already attached."

Here are the steps I took, maybe I'm doing something wrong?
1. Open open-ephys-GUI.sln and run the project
2. While that is running, open the other project *myPluginName*.sln
3. Click Debug -> Attach to Process
4. Locate open-ephys.exe and click Attatch

After I do this, the error message pops up. Any suggestions? Thanks for your help!

medengineer, 2022-02-04:
Hi, 

To get around this, change step 1 above to run Debug/open-ephys.exe from a Windows File Explorer instead of the Visual Studio .sln. Then steps 2-4 should work as expected since only one debugger will be  attached.