open-ephys/plugin-GUI #343: Building a plugin that includes Boost

samjlevy, 2020-03-12:
I'm trying to build a plugin I found that requires the boost library. Specifically, it seems like Visual Studio 2019 (on Windows 10) is failing to find any of the boost files that have the extension .hpp or .ipp. To reference these files while generating during Cmake I use the line: 
`file(GLOB_RECURSE SRC_FILES LIST_DIRECTORIES false "${SOURCE_PATH}/*.cpp" "${SOURCE_PATH}/*.h" "${SOURCE_PATH}/*.hpp" "${SOURCE_PATH}/*.ipp")`
and all the files are in the source folder (I have also manually verified these files do exist. 

The specific error I see in Visual Studio will look something like this, adjusted for various hpp and ipp files: 
`2>C:\Users\Sam\Documents\GitHub\oep-installation-master\MultiBandIntegrator\MultiBandIntegrator\Source\boostAcc/boost/accumulators/accumulators.hpp(12,10): fatal error C1083: Cannot open include file: 'boost/accumulators/framework/accumulator_set.hpp': No such file or directory`
However, I can find each of these files in my solution explorer, and each of them can be found in the solution explorer. For this example hpp file, its full path is:
`C:\Users\Sam\Documents\GitHub\oep-installation-master\MultiBandIntegrator\MultiBandIntegrator\Source\boostAcc\boost\accumulators\framework\accumulator_set.hpp`
and its relative path is 
`..\Source\boostAcc\boost\accumulators\framework\accumulator_set.hpp`
cpp files from the same tree of directories don't seem to have this issue.

Other plugins have been successfully built using this same pipeline. Any ideas on what's going on here?

anjaldoshi, 2020-03-17:
Hi,
Can you send me the code for your plugin. I want to try and replicate the issue you are facing. Also, send me the link to the Boost library you are using. If you don't have the code hosted anywhere, you can email it to me at anjal@open-ephys.org

samjlevy, 2020-03-17:
The repository for the plugin is here, and its source includes the boost library:
https://github.com/mich11/multi-band-integrator

I'm generating the solution for Visual Studio 2019 using the install_windows_2019.bat from this repository (https://github.com/tne-lab/oep-installation), which has worked for me in its entirety. I added a line to the build section to grab the multi-band-integrator plugin after replicating the directory structure for the other plugins and copying over the cmake file from one of the other plugins, though I had to amend that file to include the hpp and ipp files (see my original post).

medengineer, 2020-12-15:
Plugin has been forked [here](https://github.com/open-ephys-plugins/multi-band-integrator) with boost dependencies removed.

closed 2020-12-15 by medengineer