open-ephys/plugin-GUI #346: Suggestion of having a vertical line cursor

kouichi-c-nakamura, 2020-03-20:
I'd like to suggest a vertical line cursor to OpenEphys GUI's LFP Viewer.

A horizontal movable vertical line cursor would be useful when you want to relate an event in one channel to the others by indicating the same time.

A simple use case is as below:

You're running probe recording coupled with optogenetics. You want to relate the onset of blue laser exposure with spiking activities in the high-pass-filtered (>300 Hz) probe recordings. However, because you have many channels it is difficult to see both the TTL pulse shown in an ADC channel and all the unit channels from the probe at the same time.

jsiegle, 2020-03-20:
This is a good idea! I just added this to our "LFP Viewer enhancements" project.

kouichi-c-nakamura, 2020-03-20:
Hi, I'm glad that you like it.