open-ephys/plugin-GUI #347: Request of ability to save channel colouring scheme in a configuation XML file

kouichi-c-nakamura, 2020-03-20:
Open Ephys GUI's LFP Viewer has the functionality to assign colors to channels. For example, when you are using a probe with 4 shanks, you may want to color them according to shanks. However, as far as I am aware, this coloring cannot be saved in the configuration file.

I believe this must be pretty simple to implement, and it would be nice if it is available.

anjaldoshi, 2020-03-20:
I assume you are talking about the color grouping option int the LFP Viewer (see image below)

[image]

If you have "Reload on startup" enabled in the File menu option, then the GUI will save the option selected for color grouping in the _lastConfig.xml_ when you close it. So, it should apply the same color grouping when you run the GUI next time.

If you are talking about something else, can you please be more clear?

kouichi-c-nakamura, 2020-03-20:
Yes, that's what I meant. Thanks a lot. I'm not using "Reload on startup", but when I tried to save the configuration and manually loaded it, the color is ignored. I won't have access to the rig for a long time from now, but will investigate it.

jsiegle, 2021-05-04:
The "color grouping" setting is now part of the main options interface at the bottom of the LFP Viewer, and is always re-loaded from saved configurations.

closed 2021-05-04 by jsiegle