open-ephys/plugin-GUI #349: Noticed exactly 4096 bits missing from several channels?

reinderdorman, 2020-04-08:
Hi all,

I've noticed that I have several recordings (from last december, latest open ephys GUI at the time, lubuntu 16.04, open ephys acquisition system) who seem to miss a few chunks for data from the file.
It does not happen on all recordings at the time, only a few, and in the affected recordings there seems to be a random set of channels that has exactly 4096 bits less than the other channels. 

I think this should be salvagable by just checking for the timestamps on each recordings, and I have to find out if this is consequently the last few bits or not (I hope so, which would make it a bit easy).

my questions would be: Anyone else seen this behavior? Is it a known issue? The fact that it's exactly 4096 all the time seems coincidental (althought I thought each record had a size of 2082 bits?)

lachioma, 2020-10-05:
Hello

I am currently experiencing this issue. I am using Open Ephys 0.5.1 and the Neuropixels plugin.
In all recordings, the timestamps have 4096 values less than the data.
The same does not happen with Open Ephys 0.4.6 on the same system/PC.

Alessandro

medengineer, 2020-10-05:
@lachioma, assuming you are using the Binary recording format, can you check the end of any one of the .continuous files to see if the last 4096 values are just 0s? There is a known issue that a record node can get its 'stop' signal after the preceding processor gets its stop signal, resulting in an extra block of 0 data. Regardless, the resulting n timestamps should still map to the first n values in the continuous data.

@reinderdorman, can you confirm if this was the last few bits missing vs.  a chunk somewhere in the middle? Sounds like you were on version 0.4.5 when you first ran into this.

lachioma, 2020-10-06:
The last values of the .continuous files are in fact 0s but they are always less than 4096.

For example, in this ~2.5 sec recording, the last chunk of 0s is 3133 values long, while the timestamps have 4096 values less than the data.

[image]

markschatza, 2020-10-12:
Hello,

Adding to this to say that timestamps we are getting from the record node is skipping in the data. Attached plot shows the difference between ts(i) and ts(i-1) and it shows time between is not consistent across the data. We are using the neuropixel-pxi plugin. 

[image]

medengineer, 2020-10-12:
Looks like the timestamp count is ~4x smaller than what we'd expect given the dimensions of the continuous data. Do you mind posting a link to the data you're using in that example?

markschatza, 2020-10-19:
How do you normally upload that much ephys data?

We dug a little further and saw that all of the missing timestamps were recorded as 0's in the Data file.

jsiegle, 2020-10-19:
It's fine to just upload the timestamps file (e.g., to Dropbox). 

How are you reading in the file to Matlab? The reduced sample count and large jumps between samples suggests a potential data type mismatch.

markschatza, 2020-10-19:
We are using the load_open_open_binary("structure.oebin" , "continuous", 1) function.

lachioma, 2020-10-21:
> The last values of the .continuous files are in fact 0s but they are always less than 4096.
> 
> For example, in this ~2.5 sec recording, the last chunk of 0s is 3133 values long, while the timestamps have 4096 values less than the data.
> 
> [image]

I get the same issue on the new release of the Open Ephys GUI 0.5.2. 

In the attached example, the zeros at the end of the trace are 237. The Timestamps are 4096 less values than Data.
[image]
[image]

jsiegle, 2020-10-26:
Hi @markschatza — we found an uninitialized value in the RecordNode that we believe is the cause of the problems you were seeing. We also fixed the 4096-sample discrepancy between the timestamps and continuous data. Can you try another recording with the latest version of the GUI (https://github.com/open-ephys/plugin-GUI/releases/tag/v0.5.2.1) and see if that solves the issue?

linussun, 2020-11-12:
Hi @jsiegle ,

Just using a regular 32 channel probe here using 0.5.2 saving the **binary format**. I'm here because of the same issue. I press the play button to turn acquisition on. Then press the REC button to start recording, then again to stop it. Waiting a minute, using REC button again to create the second recording. Using load_open_ephys_binary, I find that for both Recordings, there are 4096 missing Timestamps. Is there any idea if the Timestamps are missing at the end or beginning? 

Recording 1:
```
Header: [1×1 struct]
Timestamps: [1997400×1 int64]
Data: [40×2001496 double]
```

Recording 2:
```
Header: [1×1 struct]
Timestamps: [580800×1 int64]
Data: [40×584896 double]
```

jsiegle, 2020-11-12:
Hi @linussun , we figured out this was due to the last continuous data buffer getting written twice. The timestamps are definitely missing from the end, and the last 4096 samples of data are either repeated or zero.

This is fixed in the latest version of the GUI, so please upgrade when you get a chance!

linussun, 2020-11-19:
Thanks will do!

lachioma, 2020-11-23:
I confirm that this issue has been fixed in the latest version of the GUI (https://github.com/open-ephys/plugin-GUI/releases/tag/v0.5.2.2).

Many thanks!

closed 2020-12-15 by medengineer