open-ephys/plugin-GUI #350: Missing timestamps from continuous recordings, binary format

leoreh, 2020-04-13:
Hi,

I encountered the following problem and any help will be very much appreciated.
 
The first recording in most (but not all) experiments have more data points than timestamps. This is true for continuous but not event streams. The excess data points appear on all channels (including auxiliary) and their value is zero. Thus, it looks as if the data is zero-padded. The number of zeros varies between experiments, from 241 to 139249, and they always occur within the first second of the recording ([zeroPaddingData.pdf](https://github.com/open-ephys/plugin-GUI/files/4468901/zeroPaddingData.pdf)). To clarify, as far as I could tell this only occurs after I start acquisition (press play) but not between recordings. 

I am using GUI v0.4.5 on Windows 10, i5 6-core machine w/ 32 GB RAM and a GTX 1650 GPU. 
The acquisition board is connected to a mouse shuttleDrive via the Intan 32ch headstage, and to an I/O box via its digital input (receives TTLs from a camera and stimulator). 
Data is saved as flat binary.

aacuevas, 2020-04-13:
Hi,

Are the timestamps continuous or is there a gap in the sequence?

leoreh, 2020-04-13:
Hi Aarón,

Thank you for your rapid response.
The timestamps are not continuous but rather contain a gap that corresponds to the excess data points (i.e. "zero-padding").

aacuevas, 2020-04-15:
That usually indicates a gap in received data. You said it only happens in the first recording. Does that happen when you hit play and then record, when acquisition is stopped and you start it by pressing record or it does not matter?

Have you noticed if the cpu meter gets too high? What kind of hard drive do you have? With your specs it would be rare to have an issue related to this, nonetheless.

zsong30, 2020-11-23:
Hi,  I am experiencing a smiliar issue.  The timestamps are skipping with correspondent voltage data being zero.  I plotted the differences between ajacent timestamps and attached the plot here.  The x axis is timestamp and the y axis is the difference (that should have been 1).  Does anyone know why this is happeninng?  
[image]

jsiegle, 2020-11-23:
Hi @zsong30, this is due to a bug that has been fixed in the latest version of the GUI. Can you try upgrading and see if that fixes the problem?

jsiegle, 2022-07-30:
The Binary Format should be more reliable in the latest version of the GUI. Also, as mentioned above, when recording many channels simultaneously, writing to an SSD is critical.

If you encounter any issues with timestamps files in GUI version 0.6.X, please open a new issue.

closed 2022-07-30 by jsiegle