open-ephys/plugin-GUI #352: build directory empty

robinmeier, 2020-04-20:
as precompiled binaries work i'm trying to compile plugin-GUI for mac os 10.13 from source with instructions from here:
https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/491555/macOS

it says "CMake must be invoked from the Build subdirectory"

however, Build subdirectory is empty:
https://github.com/open-ephys/plugin-GUI/tree/master/Build

instructions unclear. please help?
thanks

robinmeier, 2020-04-20:
precise error is:

CMake Error: The source directory "/Applications/plugin-GUI/Build" does not appear to contain CMakeLists.txt.

medengineer, 2020-04-20:
The Build subdirectory should initially be empty as the generated files are platform dependent. 

Given your error, you are likely missing the trailing '..' in the following line:

`cmake -G 'Xcode' ..`

robinmeier, 2020-04-20:
ah yes sorry - thank you for taking the time to point this out.

closed 2020-04-20 by robinmeier