open-ephys/plugin-GUI #359: Build Issue`

macayladonegan, 2020-06-11:
after using:
cmake -G "Visual Studio 12 2013" in the  \Documents\GitHub\plugin-GUI\Builds\VisualStudio2013 folder

I keep getting this error:
CMake Error: The source directory "C:/Users/Macayla/Documents/GitHub/plugin-GUI/Builds/VisualStudio2012" does not appear to contain CMakeLists.txt.

any advice?

medengineer, 2020-06-11:
After cloning the plugin-GUI, you should have an empty Build directory. Open a terminal in the Build directory and use the command 

`cmake -G "Visual Studio 12 2013 Win64" ..` 

This tells cmake to generate the build files for Visual Studio 12 in the Build directory using the CMakeLists.txt in the parent (..) directory.

macayladonegan, 2020-06-11:
hmm ok I don't have the empty build directory, any idea what to do about that?

medengineer, 2020-06-11:
Looks like you’re using a much older version of the plugin-GUI since you have the old Builds directory. I would either do a ‘git pull’ to get the most recent changes or clone a fresh copy.

closed 2020-10-05 by medengineer