open-ephys/plugin-GUI #360: No TTL output from acquisition board

ernie7334066, 2020-06-26:
Hi,

I'm trying to send out TTL from open ephys acquisition board. 
[image]

I have the I/O board connected to digital output and I was able to see the triggered event in plugin-GUI.
[image]
[image]

But nothing happened to a peripheral device when I connected the output channel via a BNC cable. I measured the voltage in the output channel and there was no change in voltage even if there was a triggered event in plugin-GUI. Any idea what the issue might be?

Thanks,
Ernie

closed 2020-06-28 by ernie7334066

jsiegle, 2020-06-30:
Hi @ernie7334066, sorry about the delayed response. Perhaps you already figured out that it's not possible to control the TTL outputs via software. This is something we'd like to implement eventually, but it hasn't happened yet.

ernie7334066, 2020-06-30:
Yes, I realized that after reading more carefully on the wiki. Thanks for
your reply!

Ernie