open-ephys/plugin-GUI #361: Periodic error. Finding OpalKelly. Corrupt data stream. Flushing FIFO. Problems with several versions of OE-plugin-GUI.

cskoven, 2020-07-29:
I had the issue first time last fall, where the LFP viewer suddenly showed data like this, when pressing "play":
[image]
When opening OE via terminal (to get more error messages) it sometimes also reports problems with finding/flushing the OpalKelly - or the rhd2000_usb3.bit-file - even though I'm pointing directly to it using the pop-up-file-selector:
[image]
[image]
[image]
[image]
[image]
It clearly isn't "proper" data collecting - and doesn't respond to me touching the electrode connectors - or moving the wires. Sometimes the data stream even look like this:
[image]

I have tried both building directly from source - and just using readily compiled releases from web page.

In December last year it suddenly worked one day - and I was able to acquire data on a few animals.

But when I opened OE again today - the issue arose again. I also tried with the most recent relase 0.4.6.0. Same issue:
[image]
[image]
[image]

Can someone please help me out - so I can record some animals?

Thank you so much!

Br,
Christian Skoven

aacuevas, 2020-07-29:
Hi,

The Flushing messages are part of the normal operation of the device, although they taking more than a few seconds is definitely not normal. 

Are you using the system with a headstage connected? It does not appear to be one in the GUI. What are the status of the LEDs on the acquisition board?

After a connection error appears, if you completely close the software, disconnect the board both from power and usb, then connect the board again and launch the GUI, does the same symptoms still occur?

cskoven, 2020-07-30:
Hi - and thanks for getting back to me.!

Yes, I know the FIFO flushing is a part of the normal routine. And sometimes it goes without any problems.
And that's my point. I can't figure out when and why it sometimes won't work.

I'm recording right now again - after several reboots, and powering the OE Acq. Board on/off several times.
Yesterday I repated that routine for more than 2 hours, before the Opal Kelly was successfully found - AND when it was found,  when the FIFO also flushed correctly (without hanging for 15-45min - after which i restarted). It just seems very weird and non-consistent.

I am not using a head stage. I'm using the I/O board (8 BNCs via HDMI) connected to a regular 4Ch bio-amplifier. It has been working fine for long time. But once in a while I'll spend several hours to try to get it to work.

I'll report back, when it fails again, what the LED status is.

I'm using an active USB 3.0 extension cable, connected to the USB3.0 cable - also right now (where it actually works). So I don't assume this would be the problem.

Any insight is welcome. Also maybe from people having experienced similar issues.

Thanks again.

aacuevas, 2020-07-31:
When you encounter the issue, could you check with dmesg if there is any usb-related errors? From the screenshots and your description, it seems that sometimes, for some reason, USB connection between the GUI and the device fails, and it might be a low-level issue. Maybe dmesg can offer some insight.

jsiegle, 2022-07-30:
The plugin for interfacing with the acquisition board now lives here: https://github.com/open-ephys-plugins/rhythm-plugins

If you're still experiencing this problem with the latest version of the GUI, please open an issue in that repository.

closed 2022-07-30 by jsiegle