open-ephys/plugin-GUI #363: Two channels wrote into one file. "Found corrupted record...searching for record marker"

dsdanielko, 2020-08-24:
So I was running an experiment and didn't think to check the output in a test recording before the actual experiment (stupid I know), and it turns out that one of the data channels wrote itself to the ADC1.continuous file at the same time as the actual ADC1 channel, resulting in a file that is twice the size as all the other channel recordings.

Now I am getting "Found corrupted record...searching for record marker" when I try and extract the data from the file. Is there a way to recover this file? And how do I get the data channel to stop writing into the ADC1.continuous file? There doesn't seem to be a way to choose which file each channel writes to - the little label next to the data channel in LFP viewer says ADC1 (even though it shows ephys data), as does the actual ADC1 channel.

I am writing into Open Ephys format for backwards compatibility with in-house systems.

Thanks!

aacuevas, 2020-08-24:
Unfotunately, there is no easy way to unmangle the data. It might be possible which much effort, since they both come from the same hardware, so there are some assumptions that can be done, but is far from trivial.

The open ephys format names files based on the channel names. If you open the Rhythm node panel (same button as the LFP viewer) you can see the channels names. I find it weird that it renamed a neural channel as ADC. Can you please check the names in said panel?
Another option is to open the open ephys format options and check "Rename files based on channel order", which will ignore the names and just index the channels by position, without distinction between ADC and headstage channels.

dsdanielko, 2020-08-25:
Ah, that sucks, I thought it would be a sequential write of some sort. I cannot seem to recreate the issue, i think it might be to do with having multiple headstages being swapped in and out potentially?

closed 2022-07-30 by jsiegle