open-ephys/plugin-GUI #365: Buil issue with master branch with a fresh kde neon 20.4

samuelgarcia, 2020-08-28:
Hi all,

I am not able to build today with a fresh unbuntu 20.4 like distro (kde neon in fact).

It was building and worroking perfectly with ubuntu 18.04.
It seems to be releated to Juice compilation.
Any idea ?

Thanks a lot for help.

```bash
Scanning dependencies of target open-ephys
[  1%] Building CXX object CMakeFiles/open-ephys.dir/JuceLibraryCode/BinaryData.cpp.o
[  1%] Building CXX object CMakeFiles/open-ephys.dir/JuceLibraryCode/juce_audio_basics.cpp.o
[  2%] Building CXX object CMakeFiles/open-ephys.dir/JuceLibraryCode/juce_audio_devices.cpp.o
[  2%] Building CXX object CMakeFiles/open-ephys.dir/JuceLibraryCode/juce_audio_formats.cpp.o
[  3%] Building CXX object CMakeFiles/open-ephys.dir/JuceLibraryCode/juce_audio_processors.cpp.o
In file included from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/juce_graphics.h:109,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_gui_basics/juce_gui_basics.h:55,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_audio_processors/juce_audio_processors.h:54,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_audio_processors/juce_audio_processors.cpp:36,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/juce_audio_processors.cpp:9:
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/colour/juce_PixelFormats.h: In member function ‘juce::uint8& juce::PixelARGB::getAlpha()’:
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/colour/juce_PixelFormats.h:113:77: error: cannot bind packed field ‘((juce::PixelARGB*)this)->juce::PixelARGB::<anonymous>.juce::PixelARGB::<unnamed union>::comps[3]’ to ‘juce::uint8&’ {aka ‘unsigned char&’}
  113 |     forcedinline uint8& getAlpha() noexcept           { return comps [indexA]; }
      |                                                                ~~~~~~~~~~~~~^
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/colour/juce_PixelFormats.h: In member function ‘juce::uint8& juce::PixelARGB::getRed()’:
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/colour/juce_PixelFormats.h:114:77: error: cannot bind packed field ‘((juce::PixelARGB*)this)->juce::PixelARGB::<anonymous>.juce::PixelARGB::<unnamed union>::comps[2]’ to ‘juce::uint8&’ {aka ‘unsigned char&’}
  114 |     forcedinline uint8& getRed() noexcept             { return comps [indexR]; }
      |                                                                ~~~~~~~~~~~~~^
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/colour/juce_PixelFormats.h: In member function ‘juce::uint8& juce::PixelARGB::getGreen()’:
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/colour/juce_PixelFormats.h:115:77: error: cannot bind packed field ‘((juce::PixelARGB*)this)->juce::PixelARGB::<anonymous>.juce::PixelARGB::<unnamed union>::comps[1]’ to ‘juce::uint8&’ {aka ‘unsigned char&’}
  115 |     forcedinline uint8& getGreen() noexcept           { return comps [indexG]; }
      |                                                                ~~~~~~~~~~~~~^
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/colour/juce_PixelFormats.h: In member function ‘juce::uint8& juce::PixelARGB::getBlue()’:
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/colour/juce_PixelFormats.h:116:77: error: cannot bind packed field ‘((juce::PixelARGB*)this)->juce::PixelARGB::<anonymous>.juce::PixelARGB::<unnamed union>::comps[0]’ to ‘juce::uint8&’ {aka ‘unsigned char&’}
  116 |     forcedinline uint8& getBlue() noexcept            { return comps [indexB]; }
      |                                                                ~~~~~~~~~~~~~^
In file included from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/juce_graphics.h:131,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_gui_basics/juce_gui_basics.h:55,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_audio_processors/juce_audio_processors.h:54,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_audio_processors/juce_audio_processors.cpp:36,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/juce_audio_processors.cpp:9:
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/native/juce_RenderingHelpers.h: In instantiation of ‘void juce::RenderingHelpers::EdgeTableFillers::SolidColour<PixelType, replaceExisting>::replaceLine(juce::PixelRGB*, juce::PixelARGB, int) const [with PixelType = juce::PixelRGB; bool replaceExisting = true]’:
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/native/juce_RenderingHelpers.h:636:17:   required from ‘void juce::RenderingHelpers::EdgeTableFillers::SolidColour<PixelType, replaceExisting>::handleEdgeTableLineFull(int, int) const [with PixelType = juce::PixelRGB; bool replaceExisting = true]’
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/native/juce_RenderingHelpers.h:1889:29:   required from ‘void juce::RenderingHelpers::ClipRegions<SavedStateType>::RectangleListRegion::SubRectangleIterator::iterate(Renderer&) const [with Renderer = juce::RenderingHelpers::EdgeTableFillers::SolidColour<juce::PixelRGB, true>; SavedStateType = juce::RenderingHelpers::SoftwareRendererSavedState]’
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/native/juce_RenderingHelpers.h:1513:13:   required from ‘void juce::RenderingHelpers::EdgeTableFillers::renderSolidFill(Iterator&, const juce::Image::BitmapData&, juce::PixelARGB, bool, DestPixelType*) [with Iterator = juce::RenderingHelpers::ClipRegions<juce::RenderingHelpers::SoftwareRendererSavedState>::RectangleListRegion::SubRectangleIterator; DestPixelType = juce::PixelRGB]’
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/native/juce_RenderingHelpers.h:2555:67:   required from ‘void juce::RenderingHelpers::SoftwareRendererSavedState::fillWithSolidColour(IteratorType&, juce::PixelARGB, bool) const [with IteratorType = juce::RenderingHelpers::ClipRegions<juce::RenderingHelpers::SoftwareRendererSavedState>::RectangleListRegion::SubRectangleIterator]’
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/native/juce_RenderingHelpers.h:1814:13:   required from ‘void juce::RenderingHelpers::ClipRegions<SavedStateType>::RectangleListRegion::fillRectWithColour(SavedStateType&, const juce::Rectangle<int>&, juce::PixelARGB, bool) const [with SavedStateType = juce::RenderingHelpers::SoftwareRendererSavedState]’
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/native/juce_RenderingHelpers.h:1811:14:   required from here
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/native/juce_RenderingHelpers.h:681:34: warning: converting a packed ‘juce::PixelRGB’ pointer (alignment 1) to a ‘int’ pointer (alignment 4) may result in an unaligned pointer value [-Waddress-of-packed-member]
  681 |                             int* d = reinterpret_cast<int*> (dest);
      |                                  ^
In file included from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/juce_graphics.h:109,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_gui_basics/juce_gui_basics.h:55,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_audio_processors/juce_audio_processors.h:54,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_audio_processors/juce_audio_processors.cpp:36,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/juce_audio_processors.cpp:9:
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/colour/juce_PixelFormats.h:361:17: note: defined here
  361 | class JUCE_API  PixelRGB
      |                 ^~~~~~~~
In file included from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/juce_graphics.h:131,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_gui_basics/juce_gui_basics.h:55,
                 from /home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_audio_processors/juce_audio_processors.h:54,
[... cut ...]
/home/manip/code_ephys/plugin-GUI/JuceLibraryCode/modules/juce_graphics/colour/juce_PixelFormats.h:361:17: note: defined here
  361 | class JUCE_API  PixelRGB
      |                 ^~~~~~~~
make[2]: *** [CMakeFiles/open-ephys.dir/build.make:115 : CMakeFiles/open-ephys.dir/JuceLibraryCode/juce_audio_processors.cpp.o] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:1065 : CMakeFiles/open-ephys.dir/all] Erreur 2
make: *** [Makefile:84 : all] Erreur 2
manip@openephys-maxime:~/code_ephys/plugin-GUI/Build$ 
```

anjaldoshi, 2020-08-28:
Hey Samuel,

Current version of JUCE shipped with the GUI only supports GCC version 8 or lower. As you are working with KDE Neon 20.04, the GCC version on your machine might be 9 or 10. You can check that by running `gcc --version` command in your terminal.

If that is the case, then you must install gcc-8 and g++-8 by running `sudo apt-get install gcc-8 g++-8`
Then, before running cmake, change gcc version to 8 by running the following command:
`export CC=gcc-8 && export CXX=g++-8`

This should resolve your issue. Let me know if you need help with anything else.

samuelgarcia, 2020-08-31:
Oh!
Sorry I miss this.
Thanks a lot.

closed 2020-08-31 by samuelgarcia