open-ephys/plugin-GUI #366: NpxExtractor

macayladonegan, 2020-08-28:
Hi all,

I downloaded the executables for the gui, but it seems to be missing the tools folder? Is there another way to get a hold of NpxExtractor.exe?

Best,
Macayla

jsiegle, 2020-08-28:
It's no longer included in the binary release, but you can download it here: https://github.com/open-ephys-GUI-binaries/open-ephys/tree/neuropix/Tools/NpxExtractor

macayladonegan, 2020-08-30:
Does this work with the .npx2 data format? I tried running this but nothing showed up in my output directory

jsiegle, 2020-08-31:
It should work with .npx2 files. It can be slow to start, as it searches through the entire file to count the number of probes in the recording. But if it seemed like it completed without writing anything, then there's something wrong.

I added the NpxExtractor executable to the `ecephys_spike_sorting` repo, since we're no longer updating `open-ephys-gui-binaries`: https://github.com/AllenInstitute/ecephys_spike_sorting/tree/master/ecephys_spike_sorting/modules/extract_from_npx/NpxExtractor/Release

Let me know if that one also doesn't work, and we can continue to troubleshoot.

jsiegle, 2020-10-20:
Hi there! Did you ever get a chance to test this out?

macayladonegan, 2020-10-20:
Yes it ended up working thanks!

closed 2020-10-20 by jsiegle