open-ephys/plugin-GUI #371: Pulse pal not found on COM11+

mich11, 2020-09-17:
Hi,
I noticed that when adding the pulse pal plugin to the to the signal chain, open-ephys won't find the pulse pal if it's on COM11 or higher (Windows 10, 64bit).  The pulse pal tab reads Pulse Pal: NOT CONNECTED, and open-ephys usually crashes shortly after. 

Running open-ephys with the console prints the following when I try to add it:

Sinks::Pulse Pal (1-10)
Searching for Pulse Pal...
ofSerial: unable to open port...Device not found

My guess is that open-ephys is only checking the first 10 COM ports for the pulse pal. I was able to find a different USB port to plug the pulse pal into to change its associated COM port and resolve the problem, but it would be nice to fix for future users.

jsiegle, 2020-09-17:
Thanks, this is helpful to know! We're planning to migrate the Pulse Pal module to the Plugin Installer soon (so it can be upgraded independently of the main GUI), and we'll include this fix when we do.

jsiegle, 2022-07-30:
The source code for the Pulse Pal Output now lives here: https://github.com/open-ephys-plugins/pulse-pal-output

This problem should be fixed now, but if not, please open an issue in that repository.

closed 2022-07-30 by jsiegle