open-ephys/plugin-GUI #375: NetworkEvents not communicating with Record Node

lshaheen, 2020-10-03:
Hi,

I'm trying to get this new version (0.5.1) working and am running into some trouble using the NetworkEvents plugin. 

1) Our stimulus control software sends a zeroMQ message to start recording: "StartRecord CreateNewDir=1 RecDir=XXX PrependText=YYY" Those all get set in the global recording panel, but the Record Node's directory doesn't change. Data gets saved to whatever that was set to (but it does get saved with the right prefix). Is there any way to get it to set the directory?

2) To get and save the recording path to our dataset, we send the command 'getRecordingPath' This now is returning an empty string. I think this is related to the new Record Node? Maybe since there can be multiple of these in different paths this command doesn't make sense anymore? But still the timestamp should be common to all of them, right? That's really what we need since we're setting the rest of the path above.

3) It seems like every other message is missed now. On the bottom of the OEP GUI only every other message seems to come though. And if it's a message that should get a response every other one instead returns 'failed waiting for a reply'. But bizarrely the messages that show up on the OEP GUI return 'failed waiting for a reply' and the ones that don't get a response.

medengineer, 2020-10-05:
Hi, thanks for testing this out!

1.	The RecDir param sets the global record path, which in turn, sets the path for any future record node introduced into the signal chain. We decided this was preferable to applying the new path to all current record nodes in the signal chain. Currently there is no way to set the path of an existing record node via NetworkEvents. I suggest we introduce an additional param such as ‘nodeID’ where we can specify which record node(s) to update with the desired path. Any preference on your end that would make your experiments run smoother? 

2.	The ‘getRecordingPath’ should still return the path in the Control Panel, so that is a bug on our end. Here we should also add a param such as ‘nodeID’ to retrieve the recording directory for a particular record node. If no nodeID param is included, it would return the global recording path as shown in the Control Panel.

3.	I haven’t been able to reproduce a ‘failed waiting for a reply’ message on my end. What messages are you sending over that are not getting a response? TTLs?

jsiegle, 2022-07-30:
We now recommend using the [Open Ephys HTTP Server](https://open-ephys.github.io/gui-docs/User-Manual/Whats-new-in-version-06X.html#built-in-http-server) to get and set information related to recording. If you're still experiencing problems with this approach, please open a new issue.

closed 2022-07-30 by jsiegle