open-ephys/plugin-GUI #377: Issues with splitters and mergers

aacuevas, 2020-10-06:
In 0.5.1, loading a signal chain with splitter and/or mergers returns a mangled chain with most of the processors merged into a single branch. 

In addition, sometimes when dragging a splitter and then another processor, it ends on the second branch, instead of the intended current, first branch.

I have also encountered some crashes when trying to connect a merger to a different branch. I'll further test this.

jsiegle, 2020-10-10:
We figured out the cause of this issue...saving the signal chain calls `Splitter::switchIO()`, which means that updating the recovery config after adding each processor can change the state of the underlying signal chain. 

We will release a fix for this in the next few days.

jsiegle, 2020-10-12:
This is fixed in the latest development branch. Will leave this open until we merge the changes into master.

jsiegle, 2020-10-20:
This problem is fixed in v0.5.2. The installers are available [here](https://open-ephys.org/gui), or you can download the GUI as a .zip file [here](https://dl.bintray.com/open-ephys-gui/Release/).

PR #381 (2020-10-12): Fix splitter attaching first processor to second branch causing errors
* Reset the state of Splitters to the last activePath value when adding it to signal chain while saving to recoveryConfig.
* Stop saving state to recoveryConfig while loading config on startup.
* Fix a crash with Reload on startup enabled when recoveryConfig file does not exists and lastConfig does.
* Resolves #377
closed 2020-10-20 by jsiegle