open-ephys/plugin-GUI #378: Corrupt data / TokenError reading numpy file from binary format

timsainb, 2020-10-06:
Hi All,

I am unable to read several of the numpy files saved with the binary format, with the following error:

```
channels = np.load(
    block1
    / "experiment1"
    / "recording1"
    / "events"
    / "Rhythm_FPGA-113.0"
    / "TTL_1"
    / "channels.npy"
)
channels
```

Here is the error

```
TokenError                                Traceback (most recent call last)
<ipython-input-291-69e3431c57b0> in <module>
      6     / "Rhythm_FPGA-113.0"
      7     / "TTL_1"
----> 8     / "channels.npy"
      9 )
     10 channels

python3.6/site-packages/numpy/lib/npyio.py in load(file, mmap_mode, allow_pickle, fix_imports, encoding)
    451             else:
    452                 return format.read_array(fid, allow_pickle=allow_pickle,
--> 453                                          pickle_kwargs=pickle_kwargs)
    454         else:
    455             # Try a pickle

python3.6/site-packages/numpy/lib/format.py in read_array(fp, allow_pickle, pickle_kwargs)
    727     version = read_magic(fp)
    728     _check_version(version)
--> 729     shape, fortran_order, dtype = _read_array_header(fp, version)
    730     if len(shape) == 0:
    731         count = 1

python3.6/site-packages/numpy/lib/format.py in _read_array_header(fp, version)
    591     #   "fortran_order" : bool
    592     #   "descr" : dtype.descr
--> 593     header = _filter_header(header)
    594     try:
    595         d = safe_eval(header)

python3.6/site-packages/numpy/lib/format.py in _filter_header(s)
    554     # adding newline as python 2.7.5 workaround
    555     string = s + "\n"
--> 556     for token in tokenize.generate_tokens(StringIO(string).readline):
    557         token_type = token[0]
    558         token_string = token[1]

python3.6/tokenize.py in _tokenize(readline, encoding)
    600         else:                                  # continued statement
    601             if not line:
--> 602                 raise TokenError("EOF in multi-line statement", (lnum, 0))
    603             continued = 0
    604 

TokenError: ('EOF in multi-line statement', (3, 0))
```

Has anyone run into this issue? I attached one of the files. 

[channels.npy.zip](https://github.com/open-ephys/plugin-GUI/files/5335840/channels.npy.zip)

medengineer, 2020-10-06:
Looks like the header in the channels.npy file is incorrect. Can you attach the corresponding synchronized_timestamps.npy file that loads in fine?

timsainb, 2020-10-06:
> Looks like the header in the channels.npy file is incorrect. Can you attach the corresponding synchronized_timestamps.npy file that loads in fine?

It's 1.1 Gb so here is a drive link: https://drive.google.com/file/d/1qT3qalcFQjZ-44u6Me5Mj9mMvgjju1PO/view?usp=sharing

timsainb, 2020-10-06:
re header problems:

```
with open(filename, 'rb') as fhandle:
    major, minor = numpy.lib.format.read_magic(fhandle)
    print(major, minor)
    shape, fortran, dtype = numpy.lib.format.read_array_header_1_0(fhandle)
```

yields the same error:
```
1, 0
TokenError: ('EOF in multi-line statement', (3, 0))
```

With synchronized_timestamps.npy here's the header info:

```
1 0
(152428800,) False float64
```

jsiegle, 2020-10-20:
We believe this problem was occurring when the thread that handles disk writing doesn't have enough time to shut down gracefully. This should be fixed in the latest GUI version (0.5.2). We've tested a bunch of signal chain configurations and haven't observed it. Please let us know right away if this happens again.

closed 2020-12-15 by medengineer