open-ephys/plugin-GUI #379: Separating multiple continuous files by ports

leoreh, 2020-10-09:
Hi,

I am recording from two mice simultaneously. Each mouse is connected vis SPI to a different port of an Open Ephys acquisition board. Data is saved as flat binary. 
Is it possible to separate in real time the recordings from the two mice such that two continuous.dat files will be created?

Sincerely appreciated,
Leore Heim
PhD candidate, Tel Aviv University.

holmescharles, 2020-10-09:
One solution would be to use two cascaded channel maps (or really any plugin) and save only a subset of the data at those plugins. For example, I you have N channels, at the first channel map plugin, save channels 1-(N/2), then at the second plugin, save channels (N/2 + 1)-N. To access the 2 separate files, you'd have to pull them from their respective plugin folders in the recording directories, but it should do the trick.

leoreh, 2020-10-09:
Worked wonderfully!
Thank you :)

closed 2020-10-09 by leoreh