open-ephys/plugin-GUI #380: Channel map module behavior change

chris-angeloni, 2020-10-10:
We're trying to use a channel map module to premap our data before spike sorting, but it seems like the behavior of this module has changed from prior versions (I'll note we're not using the newest version, we're using v0.4.6).

In the past, remapping using this module would remap the channels in the LFP viewer AND remap channels that are recorded in the Rhythm_FPGA folder. However, it seems like (at least in v0.4.6) this module remaps the channels down stream in the signal chain, but doesn't remap the channels recorded in Rhythm_FPGA.

The only way I can get it to record these remapped channels is to set the channel map module to record them, which I guess isn't a big deal, but may require some tweaks to our processing pipeline as it saves the data in a folder named Channel_Map.

Is this also the default behavior in v0.5.0? Many thanks!

OS: Windows
GUI: v0.4.6

jsiegle, 2020-10-12:
I'm a bit puzzled by that behavior, because remapping done by the Channel Map should only affect itself and any downstream processors, not the upstream data source. This should have always been the case, so I'm not sure why the remapping is able to propagate backwards in 0.4.5.

In 0.5.0+, recording happens inside a plugin-like Record Node that needs to be explicitly dropped into the signal chain. So a signal chain of Rhythm FPGA > Channel Map > Record Node would save the reordered channels. This change was made to improve efficiency (having everything connected to one centralized Record Node was bogging down processing in very large signal chains) and flexibility (e.g., making it possible to save data across multiple disks), but it will change the directory structure because every Record Node now has its own directory. Since this will be the standard moving forward, I'd recommend upgrading to the latest version when you can—with the caveat that there are some [open](https://github.com/open-ephys/plugin-GUI/issues/378) [recording-related](https://github.com/open-ephys/plugin-GUI/issues/376) [issues](https://github.com/open-ephys/plugin-GUI/issues/349) that we will push fixes for in the next few days.

chris-angeloni, 2020-10-12:
Hmm, that makes sense, I might be mistaken then about the past behavior (I'm probably conflating post-recording channel mapping with what I am seeing in the GUI).

And thanks for explaining the new GUI functionality! We have been waiting to switch over until it was thoroughly bug-fixed.

closed 2020-10-12 by chris-angeloni