open-ephys/plugin-GUI #384: Open Ephys GUI does not open after upgrading to v 0.5.1

clausqueiroz, 2020-10-16:
We have just updated the Open Ephys GUI and although it opened a couple of times after the update (although we experienced problems loading previous acquisition scheme/blocks) it stops opening the interface entirely today. The Debug Console (black dialog window) opens, but the GUI window stays completely white. The Debug Console informs no errors. The information we got is the following:

[image]

Any suggestions? Many thanks in advance!

anjaldoshi, 2020-10-16:
Hi, 
May I ask what build type of the GUI are you using? Did you download the .zip binaries or did you use the new installer to install the GUI?

clausqueiroz, 2020-10-16:
Hi,
We used the installer file (not the ZIP). Interestingly, after I sent the
message with our problem, we installed the same file on another computer
and it worked. Is it possible to have access to the previous version of the
GUI (version 0.4)?
Many thanks for your reply and assistance.
Claudio

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

anjaldoshi, 2020-10-16:
The previous 0.4 version we have on our server is v0.4.6, which is available here (https://dl.bintray.com/open-ephys-gui/Release/open-ephys-v0.4.6-windows.zip).
Regarding the v0.5.1 issue, can you please delete everything inside `C:\ProgramData\Open Ephys` and then try running the GUI again? If you can't find the ProgramData directory, it is a hidden directory. You can make it visible by following [these steps](https://support.microsoft.com/en-us/windows/view-hidden-files-and-folders-in-windows-10-97fbc472-c603-9d90-91d0-1166d1d9f4b5) in your C: drive.

clausqueiroz, 2020-10-16:
Great, it worked! Many thanks! Also for the link for the early version!
Best, Claudio

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

anjaldoshi, 2020-10-16:
Awesome! You're welcome! We have a fix for this issue on our development branch already, it will be in the next release pretty soon. Thanks for reporting such critical bugs, helps us release fixes for such bugs faster.

clausqueiroz, 2020-10-16:
I'm sorry to bother you again, but now we are facing other issues...

1. we cannot load pre-defined (saved) blocks. The program crashes despite
making a new chain of events in the new version (or even using the restore
previous option).
2. after making a new chain of events (and adding a recording block at the
end), the program closes after 2-3 s of recording activity (but not after
started viewing).

any suggestion on what could be possible going wrong is super welcomed!

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

jsiegle, 2020-10-16:
Hi @clausqueiroz, no worries...these problems are all related to the same underlying issue, which will be fixed in the next release (hopefully out later today).

If you can send us the .xml file of the signal chain you're using, we can double-check that everything loads properly in the latest version.

jsiegle, 2020-10-20:
This problem is fixed in v0.5.2. The installers are available [here](https://open-ephys.org/gui), or you can download the GUI as a .zip file [here](https://dl.bintray.com/open-ephys-gui/Release/).

closed 2020-10-20 by jsiegle