open-ephys/plugin-GUI #385: Load Binary Continuous Files in Matlab

saman-abbaspoor, 2020-10-23:
Hi all,

I'm trying to create a data analysis pipeline using the recorded data that we're gonna have from OpenEphys acquisition board. So, I started learning about how to create the signal chain and used the example DataFiles. I successfully saved the data from the Record Node. I would like to open the binary continuous files in Matlab and inspect them. However, when I use the 'load_open_ephys_binary', I get the following error:

**Error using strfind**
**PATTERN must be a string scalar or character vector.**

**Error in contains (line 36)**
**tf = ~cellfun(@isempty, strfind(s, pattern));**

**Error in matlab.lang.makeValidName>getMakeValidFcnHandle/makeValid (line 152)**
        **if any(contains(name, whitespace))**

**Error in matlab.lang.makeValidName>@(n)makeValid(n,@replaceWithUnderscore) (line 182)**
    **makeValidFcnHandle = @(n)makeValid(n, @replaceWithUnderscore);**

**Error in matlab.lang.makeValidName (line 111)**
    **names(modified) = makeValidFcnHandle(names(modified));**

**Error in matlab.internal.json.makeStructure (line 45)**
    **names = matlab.lang.makeValidName(names);**

I know the error is for processing the json file. This is the single line of code I use:

`D = load_open_ephys_binary('structure.oebin','continuous',1)`

And this is the part of the 'load_open_ephys_binary' code that I get the error:

`json=jsondecode(fileread(jsonFile));`

jsondecode can't process the char file that is fed by the fileread. I'm not sure how to fix this issue so I would appreciate if you can give me any instructions on how to deal with this.

medengineer, 2020-10-23:
Hi,

Can you please send the saved data as a zip to to pavel@open-ephys.org

saman-abbaspoor, 2020-10-23:
Hi, I have just sent you the saved data in compressed format.

medengineer, 2020-12-15:
This issue was resolved offline. Error was due to a conflicting matlab package.

closed 2020-12-15 by medengineer