open-ephys/plugin-GUI #394: Display NI-DAQ events in LFP viewer

lachioma, 2020-12-08:
Is it possible to have one LFP viewer that displays ephys data from Neuropix-PXI overlaid with events coming from NI-DAQmx?

If not possible, being able to view data from multiple sources (e.g. Neuropix-PXI and NI-DAQmx) would be a very useful feature to have.

Thank you!
Alessandro

jsiegle, 2020-12-08:
It's not possible to overlay the events directly, but you can use two separate LFP Viewers to visualize the NIDAQ events on the same timebase as the Neuropixels data. Just pop out one of the LFP Viewers into its own window, which you can place directly above or below the other one. Let me know if that works!

lachioma, 2020-12-24:
Thank you, it works (see screenshot). 
You just need to use "pause" on one of the two LFP viewers to align the vertical line cursor. Also, the cursors do not move at exactly the same speed (i.e. the clocks are different), so they have to be re-aligned every few minutes.

[image]

jsiegle, 2021-05-04:
This functionality is much improved with the latest version of the LFP Viewer. You can now split the display horizontally, so the Neuropixels and NIDAQ channels are stacked on top of one another. There is also a "sync displays" button that resets all displays to the left side of the screen.

More information can be found here: https://open-ephys.github.io/gui-docs/User-Manual/Plugins/LFP-Viewer.html

closed 2021-05-04 by jsiegle