open-ephys/plugin-GUI #397: Record Node not recording all channels

PhilNeuro, 2020-12-17:
A very strange issue (Windows 10, OE v0.5.2.2).  We use 3 FPGA modules.  The 3rd one has 128 channels connected, and they are shown correctly on the interface. These signals can be seen on a viewer later in the processing chain.  However, only 8 channels are recorded.  The record-node's 'settings.xml' file indicates that we do have 128 channels, but the associated structure.oebin only lists 8 channels, which is what we can find in the continuous.dat file.    On this setup, we have a total of 512 channels.   Is it possible that we have to fill the FPGAs  in sequence before adding channels to the next board ?
[image]

jsiegle, 2020-12-17:
That is indeed strange. Can you attach the settings.xml file to this issue? Also, can you try deselecting all of the channels in Rec 109 "SP0" interface, then re-selecting them and see if that fixes the problem?

Also, I'm interested to know what the Spike Monitor plugin does!

PhilNeuro, 2020-12-17:
Here is the 'settings.xml' file, thanks!   I just saw that the record node lists all 128 channels, but only marks 8 of them with a recordstate of '1'.   I deselected and reselected them in both the FPGA and REC 109 modules without change.
[settings.zip](https://github.com/open-ephys/plugin-GUI/files/5711346/settings.zip)

The spike monitor is my merged version of Spike Detector and Viewer where only a subset of the acquired channels are active and visible at once, needed for our high count channel system where we only store continuous data anyway.   Functional but not yet fully debugged.  
[image]

jsiegle, 2020-12-17:
Thanks! At least the settings file is consistent with the actual behavior. 

Are you using the Record Node selection interface to toggle the channels on/off, as in the attached image? The "PARAM" buttons don't have any effect in the new Record Node.

[image]

PhilNeuro, 2020-12-17:
Bingo !  I was using the PARAM button  !    I used to collect only 8 channels on this FPGA, and just added 128 without modifying the signal chain.   I assumed the 'updateSettings' calls would take of this since the PARAM section was correctly updated.   Is it possible that adding/removing signals on my FPGA does not automatically change the record-node  active inputs ?

jsiegle, 2020-12-17:
This is something we'll need to look into. The default behavior should be to record all additional channels if the number of inputs to the Record Node increases. If that's not the case right now, then we'll get that fixed.

PhilNeuro, 2020-12-17:
All right, thanks a lot, this was very helpful (and fast !)

closed 2020-12-17 by PhilNeuro