open-ephys/plugin-GUI #398: sync_messages.txt empty

yves-weissenberger, 2020-12-18:
Hi,

I'm using the open-ephys GUI to record electrophysiology signals and am using a series of TTL pulses to sync the open-ephys board with separate equipment. To do this alignment, I use the start time in the sync_messages.txt file. However, l've been looking through previous snyc_messages.txt files and noticed that many are empty, despite recording data and TTL timestamps being present. Any idea what might be causing this or how to fix this issue? Is there another place I can obtain this start time from? Many thanks!

jsiegle, 2020-12-18:
Hi Yves! Which version of the GUI are you using? And are you using the Binary or Open Ephys format? Either way, you can also get the start time from the first timestamp in the continuous data files. Does that work for synchronizing your TTLs?

yves-weissenberger, 2020-12-18:
Hi,

I am using Open-Ephys-GUI-v0.5.2.2 and saving in the Binary format. Running on windows 10. Accessing the first timestamp works great. Thanks a lot and happy holidays!

closed 2020-12-18 by yves-weissenberger