open-ephys/plugin-GUI #400: Accumulating spike waveforms

TamirEliav, 2021-01-10:
Hi
Is there a way to change the number of accumulated spikes (waveforms) in the spike viewer plugin? Can it be set to be infinite (i.e. spikes keep accumulating until the user clears the screen)
Currently the spikes waveforms appear and disappear on seconds timescales (the data points in the 2D display keep accumulating though...).
Also, is there a way to choose the voltage range to values different than 100/250/500uV?

On a different manner, the GUI is not using the system audio output device. How can I select the audio output device?

jsiegle, 2021-01-10:
If you compile the GUI yourself (instructions [here](https://open-ephys.github.io/gui-docs/Developer-Guide/Compiling-the-GUI.html)), you can change [this line](https://github.com/open-ephys/plugin-GUI/blob/343544b584d374213f5bbb96e60ea715d55b5770/Plugins/BasicSpikeDisplay/SpikeDisplayNode/SpikeDisplayCanvas.cpp#L872) to increase the `bufferSize` variable in the Spike Viewer's `WaveAxes` class. That will increase the number of spikes that are overlaid (the current value is 5).

For the audio device issue, there is an "audio settings" interface in the control panel accessed via the button that displays the current buffer size (default = 23 ms). That will allow you to select the output device you want the GUI to use, and to output a test signal to make sure it's working properly.

TamirEliav, 2021-01-13:
thanks!
I successfully compiled it with some larger buffer (100 spikes), but effectively it didn't work because it always crashed after a few seconds (using 16 tetrodes), even when using a very strong workstation.

TamirEliav, 2021-01-14:
Thanks for the advise!
I re-compiled the GUI with this change, setting the buffer size to 1000.
Strangely, It seems that now the viewer is accumulating more than 1000 waveforms: I inserted a 1kHz sinewave input, thus expecting a "waveform lifetime" of 1 sec, but it seems to appear on-screen many many more seconds (to check this, I changed the amplitude after accumulating waveforms for a few seconds, and waited until the old amplitude waveforms disappeared).

jsiegle, 2021-01-20:
It looks like the number of spikes sent to the canvas on each processing cycle is throttled by another variable, `displayBufferSize` in [SpikeDisplayNode.cpp](https://github.com/open-ephys/plugin-GUI/blob/343544b584d374213f5bbb96e60ea715d55b5770/Plugins/BasicSpikeDisplay/SpikeDisplayNode/SpikeDisplayNode.cpp#L32). That most likely explains why you're not seeing every spike added to the buffer.

closed 2022-07-30 by jsiegle