open-ephys/plugin-GUI #401: Network events not recorded

nikolaskaralis, 2021-01-10:
In all previous versions of the OpenEphys GUI, the network events sent to this node were saved automatically in the messages.events file.

However, since the change to the plugin version of the GUI, I cannot manage to make this work.

If you just add a Network Events node, commands and other messages are received normally, but are not saved.

If I try to add a second Record Node next to the Network Events node, this is not allowed by the software and acquisition is disabled.

If I merge the Network Events stream with the Rhythm FPGA stream (as suggested in the User Guide) and then add a Record Node, this again disables acquisition.

Unless I am doing something wrong, this seems to be a bug of either the Network Events or the Record Node plugin and it breaks a core functionality of the GUI (saving network messages).

Any ideas how I can overcome this issue?

jsiegle, 2021-01-10:
It looks like this was due to the NetworkEvents module not being updated to be compatible with changes introduced in GUI version 0.5.3. Once I apply the change [in this pull request](https://github.com/open-ephys-plugins/NetworkEvents/pull/3), NetworkEvents are saved as expected.

We'll release a new version of the NetworkEvents module that fixes this in the next day or so (which will be available via the Plugin Installer).

nikolaskaralis, 2021-01-10:
Hi Josh,
that's great!
I will test the new version once is it released.

anjaldoshi, 2021-01-11:
New version of NetworkEvents (0.1.1) that fixes this issue is up! You can update it via the Plugin Installer.

nikolaskaralis, 2021-02-04:
I can confirm this fixed the issue.

However, I have run into another (perhaps related) problem.

I am sending a remote command to start the recording.

Normally the first two lines of the messages.events file look like:
108917760 StartRecord CreateNewDir=1
108917760 Processor: Rhythm FPGA Id: 100 subProcessor: 0 start time: 108917760@30000Hz

However, lately, the vast majority of my recordings are missing these first two lines, but sometimes it is recorded.
I have not managed to identify the source of this issue.

Is there any way to recover this piece of information from the rest of the data?
I need this in order to properly timestamp the rest of the messages.

Thanks in advance for your help!

jsiegle, 2021-02-04:
Yes, the start time is redundant with the first timestamp in the continuous data.

closed 2022-07-30 by jsiegle