open-ephys/plugin-GUI #402: Config file location

nikolaskaralis, 2021-01-11:
Where is the last config file saved?

I ran into the following issue:
The last time the GUI was closed, it had the "Reload on startup" option selected.
For some reason, every time I tried to open the program it would crash, because it was trying to reload the last config and for some reason something was failing.
Uninstalling and reinstalling the GUI didn't solve the issue.

What eventually solved the issue was to install v0.5.2.2, which when started complained that the config was saved from a newer version. It nevertheless loaded it fine without crashing. Then I de-selected the reload on startup and now the v0.5.3 works.

So this points to an issue with config files, but also shows that the config is saved in some location that is not removed when uninstalling.

anjaldoshi, 2021-01-11:
If you built the GUI from source, config files are saved at the same level as where the GUI executable was generated after building. If you using pre-built binaries (installer/zip), then the config files are saved in `C:\ProgramData\OpenEphys`. The ProgramData directory is a hidden directory, so you'll have to enable the ["Show hidden files, folders, and drives"](https://support.microsoft.com/en-us/windows/view-hidden-files-and-folders-in-windows-10-97fbc472-c603-9d90-91d0-1166d1d9f4b5) option in File Explorer. We do not remove the config files during uninstallation as they are user-generated files and the user may prefer keeping those files after uninstalling the GUI. 

If you don't want to remove the config files, you can also turn off "Reload on startup" option via the `windowState.xml` file in the same `C:\ProgramData\OpenEphys` directory by setting the `shouldReloadOnStartup` attribute value to 0.

With your case, it looks like the config file some how saved a configuration that caused crashes in v0.5.3 and not in v0.5.2.2. Do you still have the config with you? If so, can you send it over to me at anjal@open-ephys.org so that i can take a look at why did it stop working with v0.5.3.

nikolaskaralis, 2021-01-12:
Hi Anjal,
thanks for the info. I did not know about this directory.

I sent you by email the config files.

Best,
Nikolas

closed 2021-01-12 by nikolaskaralis