open-ephys/plugin-GUI #404: Improve scatter plots of spikes (SpikeViewer)

TamirEliav, 2021-01-14:
It could be very helpful to use some transparency for the dots plotted in the SpikeViewer scatter plots.
Currently, it feels very hard to see nicely isolated clusters in this viewer - it looks like a uniform blob of white dots (that merge with adjacent clusters), instead of a 2D gaussian blob. I think adding transparency will resolve this problem and will enable seeing the gaussian shape of the cluster.

jsiegle, 2021-01-20:
This is a good suggestion, and pretty easy to implement! We will likely add a transparency option in a future release.

TamirEliav, 2021-01-24:
Great!
Can you give a quick pointer to where and how to set transparency?
We have now an implanted animal that we carry tests using the OpenEphys system, and it would be great to add and test this functionality as long as we have nice clusters.

ANickerson, 2021-01-24:
I think this is the line -

https://github.com/open-ephys/plugin-GUI/blob/be5c2e4842a3cc8b4d3e6fa5ca8ac5d0b826df75/Plugins/BasicSpikeDisplay/SpikeDisplayNode/SpikeDisplayCanvas.cpp#L1309

jsiegle, 2021-01-24:
Yeah, it's actually line 1308...just change it to `g.setColour(col.withAlpha(0.2));` (or substitute whatever alpha value looks reasonable).

closed 2025-06-13 by anjaldoshi