open-ephys/plugin-GUI #406: Trouble with KWIK plugin installation

yaronGoldberg, 2021-01-20:
Hi, 
I've been trying to install the KWIK plugin and ran into the following problem:
[image]
I am using the most updated windows version of the GUi, do you have any advice?
Thank you

anjaldoshi, 2021-01-20:
Hi,

Can you check whether KWIK format plugin's dependencies:` OpenEphysHDF5.dll`, `hdf5.fll`, `hdf5_cpp.dll`, `szip.dll`, and `zlib.dll` are installed in: `C:\ProgramData\Open Ephys\shared` after you install the plugin. The ProgramData directory is hidden, so make sure you unhide it by following [these steps](https://support.microsoft.com/en-us/windows/view-hidden-files-and-folders-in-windows-10-97fbc472-c603-9d90-91d0-1166d1d9f4b5).

yaronGoldberg, 2021-01-21:
Hi,
Thank you for the quick response,
The dependencies are all installed.

yaronGoldberg, 2021-01-25:
Any other suggestions?

anjaldoshi, 2021-01-25:
Can you try installing NWB format plugin and see if that works? I just want to make sure whether the issue is because of hdf5 dependencies or because of the plugin itself.

yaronGoldberg, 2021-01-25:
Same error with NWB.

anjaldoshi, 2021-01-25:
Can you uninstall the plugins and its dependency by removing `KWIKFormat.dll` & `NWBFormat.dll` from `C:\ProgramData\Open Ephys\plugins`, and only  `OpenEphysHDF5.dll` from  `C:\ProgramData\Open Ephys\shared` , and then re-install the plugins? See if that works.

yaronGoldberg, 2021-01-26:
No luck here - same error.

anjaldoshi, 2021-01-26:
I cannot reproduce the error on my end at all. Can you install the GUI and the plugins on a different machine and see if that works?

yaronGoldberg, 2021-01-26:
Same error pops on a different machine. I thought to mention that the path that appears in the error message (D:\a) does not exist on my computer. In addition, I wanted to clarify that the error appears when I open the gui (since it tries to load the plugins) and not only during the installation.

anjaldoshi, 2021-01-26:
The path just points to a path on the Github Actions server where the plugin was built. So, nothing to worry about over there.
One thing you can try, as suggested by @aacuevas, is to install the [VS2013 runtime](https://www.microsoft.com/en-US/download/details.aspx?id=40784) as the HDF5 DLLs are built with 2013 so they still need its runtime. If installing this runtime works, then let me know so that I can include this runtime to be installed by the installer in future versions.

yaronGoldberg, 2021-01-28:
It worked! Thank you very much for the help!

jsiegle, 2022-07-30:
The Kwik format has now been deprecated. We recommend using the [Binary Format](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/Binary-format.html) instead.

closed 2022-07-30 by jsiegle