open-ephys/plugin-GUI #407: version 0.5.3 impedance measure not working

dlayfiel, 2021-01-27:
Is anyone else having issues using the impedance measure function on the 0.5.3 version of the open ephys software, im use to the gui just not look likes its doing anything like in the previous versions but the software is literally doing nothing when I click the measure the impedance button.

jsiegle, 2021-01-27:
I'm not able to reproduce this (tested on Windows with v0.5.3). Can you provide some more details about your setup? For example, what type of headstages are you using?

sppmg, 2021-11-08:
Hi,
I got same issue in version v0.5.5.3 and v0.5.5.2.
The debug console display `Saving impedance  measurements in \impedance_measurements.xml`
[image]

And I tried output format in  `Open ephys` and `NWB`, same result.

jsiegle, 2021-11-10:
I believe the saving of impedance measurements was likely broken when we switched over to the Record Node system in v0.5.0+. This will be restored in the next major release (v0.6.0), but for now you'll just have to screenshot the Rhythm FPGA configuration window.

If someone has been successfully saving impedance measurements in recent version of the GUI, please let me know!

sppmg, 2021-11-11:
Oh...
The `screenshot` is too inefficient (we need log it to spreadsheet). Instead, I will use `v0.4.6` to save impedance file, it's work on same system.

Thanks your reply and hope new version coming soon.

jsiegle, 2022-07-30:
Impedance measurements can now be saved directly from the Acquisition Board plugin. For more information, see the [Rhythm Plugins](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Rhythm-Plugins.html) documentation page.

closed 2022-07-30 by jsiegle