open-ephys/plugin-GUI #408: Not a valid file issue  saving and reusing channel map

dlayfiel, 2021-01-27:
Hello I am using open ephys v0.5.3

I am unable to reuse channel map after closing the software I am able to create and save it but it gives a not a valid file
after trying to reuse the created channel map file.

jsiegle, 2021-01-27:
There are two ways to save the channel map settings:

(1) File > Save as... (saves an XML file with the entire signal chain, including the channel map). This one is re-loaded by selecting File > Open...

(2) Clicking the disk icon in the upper right of the Channel Map editor (saves a JSON file with only the channel map settings). This one is re-loaded by selecting the other icon in the Channel Map editor.

Can you please clarify which method you're using, and attach the file that's being saved?

jsiegle, 2022-07-30:
This should be fixed in the latest version of the GUI. If you're still experiencing this problem, please re-open this issue.

closed 2022-07-30 by jsiegle