open-ephys/plugin-GUI #409: Network events & eventbroadcaster "Loading Plugin Failed!!"

kidderk, 2021-01-27:
When trying to download the networkevents and event broadcaster modules via the Plugin installer from either GUI version 5.3 or 5.2.2 I get this error message in the Console:

Downloading Plugin: NetworkEvents...
D:\a\plugin-GUI\plugin-GUI\Source\Processors\PluginManager\PluginManager.cpp:219: Failed to load plugin DLL: DLL Error 0x7e
Loading Plugin Failed!!

I was however able to download the matlab interface module. Any ideas as to why networkevents and eventbroadcaster will not download? 

I downloaded the latest GUI directly from the open-ephys.org website.

anjaldoshi, 2021-01-27:
Regarding the issue, can you try and install the [VS2013 runtime](https://www.microsoft.com/en-us/download/details.aspx?id=40784) and see if that resolves your issues.

jsiegle, 2022-07-30:
The latest version of these plugins no longer have the VS2013 runtime dependency, so hopefully this issue is resolved.

closed 2022-07-30 by jsiegle