open-ephys/plugin-GUI #410: Сrash on startup

Alkohost, 2021-01-29:
I just upgraded to the latest version but still face with crash when launch the GUI (win7). 
It happens if 
1) the configuration is changed (if one of cables is disconnected) but option Reload on startup is enabled (in the previous version I just manually deleted previous config before start the GUI). 
2) the board power is off and I press cancel.

Is it possible to fix this crash problems with coming software updates?

jsiegle, 2021-01-29:
Thanks for reporting this! We will definitely get these fixed.

Alkohost, 2021-02-01:
Thank you. As far as I remember, there was also a crash due to bug/feature if a PC did not have an audio output connected. Is it fixed already?

Сan I ask questions about hardware here or should I write to the /acquisition-board ?

jsiegle, 2021-02-05:
This is fixed in the latest release (0.5.3.1).

Regarding the audio output, the GUI needs to be able access an audio output device, otherwise it won't be able to function (all of the data acquisition callbacks are driven by the audio card).

closed 2021-02-05 by jsiegle