open-ephys/plugin-GUI #413: GUI crashes when loading settings with Rhythm FPGA plugin

ldclaar, 2021-02-09:
Hello,

I am using Open Ephys, version 0.5.3, with Windows 10. I have noticed that the GUI crashes if I try to open a saved settings file that contains the Rhythm FPGA source plugin. This happens even though the Open Ephys acquisition board and headstage are connected (and recognized if I add the plugin manually).
[image]

It also crashes if I try to reopen the GUI if the Rhythm FPGA plugin was in use prior to closing the GUI. It continues to crash until I delete the lastConfig.xml file. After deleting that file, the GUI opens successfully.
[image]

I don't have the problem with the Neuropixels PXI plugin though, which is the only other plugin I have been using.

Thanks,
Leslie

jsiegle, 2021-02-09:
Hi Leslie,

This is fixed in the latest release—if you upgrade to [v0.5.3.1](https://github.com/open-ephys/plugin-GUI/releases/tag/v0.5.3.1) the config files should load again.

ldclaar, 2021-03-01:
Indeed, they do! Thanks!

closed 2021-03-01 by ldclaar