open-ephys/plugin-GUI #415: Warning: timestamps and nsamples are different when loading raw binary recording with python

shirquinn, 2021-02-21:
replicating the issue for @jsiegle request
**prior to recording I unchecked ADC channels 34-40 from the recording node channel selector interface**

>>> recording.get_channel_ids()
.dat:  continuous.dat
Estimated samples:  **44648947**  Numchan:  40
**Warning: timestamps and nsamples are different!**
[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39]
>>> timestamps_file = np.load('D:/shir_data/V1/DS/DSB4838/2021-02-16_18-47-35/Record Node 103/experiment1/recording1/continuous/Rhythm_FPGA-100.0/timestamps.npy')
>>> len(timestamps_file)
**54119936**

There is a difference of about 300 ms between the uploaded estimated samples and original timestamps file

If i don't mess with the recording node prior to recording, it doesn't produce this error

shirquinn, 2021-02-23:
Hi @jsiegle 
So i did another recording without changing anything on the default record node, and there was a discrepancy in the timestamps again

>>> recording = se.OpenEphysRecordingExtractor(recording_folder)
Loading Open-Ephys: reading settings...
Decoding data from  binary  format
Reading oebin file
D:\Anaconda3\envs\myenv\lib\site-packages\pyopenephys-1.1.3-py3.7.egg\pyopenephys\core.py:765: UserWarning: timestamps and nsamples are different!
>>> recording.get_num_frames()
72063488
>>> rec = np.load('D:\shir_data\V1\WT\Feb2021\WTB4901/2021-02-23_16-38-42\Record Node 103\experiment1/recording1\continuous\Rhythm_FPGA-100.0/timestamps.npy')
>>> len(rec)
72026367

a difference of 37,121 samples (fs = 30000)

jsiegle, 2021-02-23:
What version of the GUI are you using? Also, can you upload the `settings.xml` file that was saved along with the recording?

shirquinn, 2021-02-23:
v0.5.3.1
xml:
[https://drive.google.com/file/d/1HInz-hISW95dtTLMz2lq5qLyQi4_dz8f/view?usp=sharing](url)

shirquinn, 2021-03-07:
hi @jsiegle , I uninstalled and re-installed everything and this bug keeps happening every once in a while, any news regarding this issue?
Thanks

jsiegle, 2021-03-07:
I just tried 5 recordings with the same signal chain, and wasn't able to replicate the problem.

A few more questions:
- Do the CPU meter (yellow bar at the top of the window) or recording buffer monitor (visible if you click the vertical lines on the left side of the Record Node) ever approach 100%?
- Are there any jumps in the timestamp values, or do they always increment by 1?
- Do the continuous samples that are loaded look reasonable, or do they seem like they could be corrupted?

shirquinn, 2021-04-08:
hi, thanks for the reply

Do the CPU meter (yellow bar at the top of the window) or recording buffer monitor (visible if you click the vertical lines on the left side of the Record Node) ever approach 100%?
no

Are there any jumps in the timestamp values, or do they always increment by 1?
 increment by 1

Do the continuous samples that are loaded look reasonable, or do they seem like they could be corrupted?
corrupted

this bug happened with 2 recordings

shirquinn, 2021-04-08:
I've re- recorded one of these files with the file reader in the GUI, and I've noticed that during the first two minutes of the recording it dropped quite a lot of time points, after these 2 minutes the recording looks normal. 
in the new recorded file, there is no mismatch anymore between timestamps and samples.
I'm not sure why this happened, but I can live with the new recording for now.
If this happens again I will let you know, and if you want me to send this recording I will.
Thanks 

[image]

shirquinn, 2022-06-23:
[https://drive.google.com/file/d/1ZLpOYc5UyUy7dlpF8jaYwrLDa6FmxhsV/view?usp=sharing](url)

Thank you @jsiegle !

jsiegle, 2022-06-28:
Hi @shirquinn, I had a chance to look at this dataset. Although there is quite a bit of dropped data, the timing of the drops is  consistent with the gaps in the timestamps. Furthermore, the last recorded timestamp (92144639) exactly matches the number of samples in the continuous data file.

So, the true timestamps for the continuous data file are just 0-9214439, in consecutive order. The data in the gaps will not be recoverable, but you can still align + analyze the data that is intact.

Was there anything strange about this recording, e.g. CPU meter spiking or record buffer overruns? And are you writing data to an SSD? It seems like the computer was not able to keep up with the data throughput for some reason.

In the next major release (0.6.0), we have implemented a number of changes to the Record Node that will prevent these drops from occurring. Be on the lookout for an announcement about that in the next few weeks.

jsiegle, 2022-07-13:
Writing to an SSD is strongly recommended, so it's good that you're doing that!

It sounds like Event Triggered Avg node could be causing these issues. I would recommend removing that from the signal chain if you haven't already.

In the next release of the GUI, this plugin will be replaced with the [Online PSTH](https://github.com/open-ephys-plugins/online-psth) plugin, which has been re-written from the ground up to be much more stable and user friendly. Hopefully that will solve the issue in the future!

jsiegle, 2022-07-30:
GUI v0.6.0 has now been released. If you experience this problem with the latest version, please open a new issue.

closed 2022-07-30 by jsiegle