open-ephys/plugin-GUI #416: Spike Viewer Visualizations - Signal Chain Tutorial

cgaletzka, 2021-02-23:
I experience some troubles with the spike viewer module in the most recent release. I followed the tutorial for building a signal chain on a MacBook as described here: https://open-ephys.github.io/gui-docs/User-Manual/index.html

There is something wrong with the visualizations shown in the viewport: 
[image]

I also did the the tutorial with the older GUI on a windows PC. Here, the visualizations look fine: 
[image]

I'm not sure if this difference is caused by a setting in the GUI or due to the hardware. The signals look fine when viewed in the LFP Viewer. Help would be greatly appreciated!

jsiegle, 2021-02-23:
Hmm...I'm not seeing this on my iMac or Mac laptop. What version of macOS are you using? Any chance you have an M1 chip?

cgaletzka, 2021-02-24:
Sorry I haven't mentioned the specs. I'm using macOC 11.0.1 Big Sur on a 2.3 GHz Dual-Core Intel i5 with 16 GB RAM. The issue also persists after reinstalling the GUI.

jsiegle, 2021-02-24:
Ok, looks like this is a known issue with Big Sur: https://forum.juce.com/t/apparent-serious-juce-painting-issue-in-big-sur/43430/33

Our next big project (~3 month timescale) is to upgrade to the latest version of JUCE, which should solve the problem (at least according to the linked thread). Unfortunately, I don't think we'll be able to fix it before then.

cgaletzka, 2021-02-24:
Okay, thank you for your help!

jsiegle, 2021-02-24:
How does the LFP Viewer look? Is that usable?

cgaletzka, 2021-02-24:
Yes, the LFP Viewer works fine.

jsiegle, 2022-07-30:
The GUI now uses version 6 of the JUCE library, which solves this problem.

closed 2022-07-30 by jsiegle