open-ephys/plugin-GUI #423: GUI hanging

hwpdalgleish, 2021-03-25:
Hi,

I've been using OpenEphys with no issues for a couple of weeks installed via the installer here:
https://open-ephys.org/gui

Today when I tried to open the GUI, the console opens but then hangs at "Got creator" (see attached screenshot) and the GUI window opens in the Windows taskbar, but never opens on the monitor display (the thumbnail in the taskbar is blank). I have tried all the basic stuff - shutdown, restart, uninstalling and reinstalling the GUI several times - however this keeps happening.

[image]

One thing I have noticed is that even though I've installed a new version (i.e. should only have the default plugins) when trying to open the newly installed GUI the console still reports loading plugins that I previously downloaded via the plugin installer., i.e. "Loading Plugin: probe-viewer" in the above screenshot. It's my understanding that these plugins shouldn't yet be installed. Is there some metadata that is left after uninstalling that the newly installed version could be referencing when it tries to open that is causing the problem? Any help would be much appreciated!

Thanks,
Henry

jsiegle, 2021-03-25:
There are some supporting files located in `C:\ProgramData\Open Ephys`, including the DLLs for installed plugins and the last used configuration. These are not overwritten when updating the core application, so you don't lose your plugins and settings. But if you want to start fresh, it's fine to delete the contents of that directory.

I would recommend at least removing `lastConfig.xml` and `recoveryConfig.xml`, and see if that fixes the problem.

hwpdalgleish, 2021-03-26:
That makes sense. In the meantime compiling the GUI from source seemed to fix the problem (which I needed to do anyway to install some external plugins), but I've deleted the files you mentioned just in case I re-download the pre-compile version in future. Thanks anyway!

closed 2022-07-31 by jsiegle