open-ephys/plugin-GUI #430: A question regarding channel map before recording node

shirquinn, 2021-04-08:
Hi,
If I place the recording node **after** the channel map, how does it actually save the data?
if for example the map in the rhythm FPGA recording was 17 20 4 7
if I run the gui again with the file reader using the oebin file, and insert a channel map to the chain - will the above channels be registered now as 1 2 3 4? or do i need to upload the previous channel map again?

I hope this question was clear enough
Thanks

jsiegle, 2021-04-08:
The channels will be reordered in the recorded data. So if 17 is set as the first channel in the Channel Map, it will become channel 1 when the data is read back in.

shirquinn, 2021-04-08:
Thank you for the quick reply

closed 2021-04-08 by shirquinn