open-ephys/plugin-GUI #432: GUI crashes when loading Intan Rec. Controller with ADC 1-8 toggle on

biomath, 2021-04-12:
Dear developers,
I had been using and enjoying the OpenEphys GUI for a couple of years using the Intan Interface Board, but in my current lab we have an Intan Recording Controller. 

I have been frustrated by constant sudden crashes of the GUI when loading previously tested and working configuration files. After quite some time puzzling over why my GUI was crashing, I narrowed the issue down to reliably happening when **loading the "Intan Rec. Controller" source when the ADC 1-8 button is active**. 

I can save a config file with that button toggled off, and it loads fine. Toggling that button on after that works great. The issue only happens when loading a config that had been saved with that button on.

Fixing this bug would save me a lot of headache in having to set up channel maps every time I load a configuration.
Thanks!

jsiegle, 2021-04-12:
Thanks for reporting this! This likely has the same underlying cause as issue #413, just applied to the Intan Recording Controller. We'll get this fixed and send you some updated binaries to test (hopefully today or tomorrow).

jsiegle, 2021-04-13:
Are you using Linux, Mac, or Windows?

biomath, 2021-04-14:
> Are you using Linux, Mac, or Windows?

Sorry, I should have added more details. 
Windows 10. GUI v0.5.3.1.

jsiegle, 2021-04-14:
Please unzip the file below and use it to replace the existing `IntanRecordingController.dll` in the "plugins" directory of your Open Ephys executable. Let me know if that fixes the crash on reload...unfortunately we don't have the Recording Controller hardware available for testing at the moment.

[IntanRecordingController.zip](https://github.com/open-ephys/plugin-GUI/files/6313934/IntanRecordingController.zip)

biomath, 2021-04-16:
Working great so far! Thank you!

closed 2021-04-16 by biomath