open-ephys/plugin-GUI #434: NetworkEvents StartRecord RecDir option not updating individual network nodes

aacuevas, 2021-04-19:
Starting recording through a ZMQ command using the RecDir option updates the ControlPanel (default) path but does not update the individual Record Node paths, resulting in recording being written to the original path.

arnefmeyer, 2021-04-19:
@aacuevas I had the same issue so I added a function to CoreServices.cpp to change recording paths:
https://github.com/arnefmeyer/plugin-GUI/commit/cf6f778dfedeb7c8a4c0ae7206d15d5f714dcd07

It also requires small modifications to the NetworkEvents plugin code:
https://github.com/arnefmeyer/NetworkEvents/commit/1e3e373dd5a08ba3f7638b390fe08250581538fa

Two more things that might need to be done:
- a way to set recording paths for specific recording nodes, e.g., RecordNode=102 RecordNodeDir=dir1 RecordNode=103 RecordNodeDir=dir2 ...
- the path name in the RecordNode plugin editor is currently not updated

aacuevas, 2021-04-20:
Hi Arne,

We are going to do basically the samet, but just modifying the normal CoreServices::setRecordingDirectory instead of ceating a new method for now and without modifying the network events plugin. We will implement a proper selector for specific nodes in the future, but that requires modifying the plugin API (as you did) and this is something we want to avoid until the next major release.

Thanks for pointing out that the path in the individual nodes editors are not updated! I will take that into account.

aacuevas, 2021-04-21:
Temporary fix is already in development branch.
Calling CoreServices::setRecordingDirectory(), as NetworkEvents does, will update all record nodes in the chain. (Manually changing the default path in the control bar will not, as it is the case now).

Keeping the issue opened until a more advanced solution requiring an API change, including selection of individual node paths, is implemented.

stinos, 2021-08-25:
When changing the API, would there be an option to make appending of `RecordNode::generateDirectoryName()` optional? I.e. when calling `CoreServices::setRecordingDirectory('x')` it would be nice to have it such that the record node data ends up directly in x: it's fairly common that software driving experiments already creates its own directory structure, so forcing the data to be in an extra subdirectory is unnecessary in that case and requires extra directory scanning when loading the data again.

jsiegle, 2021-08-25:
Yes, we can definitely add this! @medengineer, can you add this option to the Record Node?

jsiegle, 2022-07-30:
The option to use a fixed directory name (rather than the auto-generated date string) is available in GUI v0.6.X. See [the documentation](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/index.html#global-recording-options) for more details.

closed 2022-07-30 by jsiegle