open-ephys/plugin-GUI #436: Processors "jump" to another stream when using 2 splitters

biomath, 2021-04-21:
Hi,

I'm running v0.5.4 (this issue was also present on 0.5.3) on Windows 10 with the Intan Rec. Controller.

When I save a configuration (attached) using 2 splitters (screenshot 1), after reloading the same configuration, processors on one side of the second splitter "jump" to the end of the first splitter  (screenshot 2). This happens regardless of what kind of processor I put on that end.

Let me know if you need any more information/files.

Thanks!

Screenshot 1:
[image]

Screenshot 2:
[image]

Config file:
[MML_NN_A4x16-Poly2-5mm-20s-lin-160_054.zip](https://github.com/open-ephys/plugin-GUI/files/6353252/MML_NN_A4x16-Poly2-5mm-20s-lin-160_054.zip)

jsiegle, 2021-04-21:
Thanks, this is helpful! We'll get this fixed before the next release.

jsiegle, 2021-04-25:
Figured out what was causing this...when selecting a processor with an upstream splitter, the splitters were not getting updated to point toward the correct branch of the signal chain. This was also making it so selecting a processor in the Graph View didn't necessarily make its editor visible.

After fixing this, the signal chain in your example re-loads fine (as well as some more complex signal chains I tested).

PR #439 (2021-04-29): 0.5.4 pre-release
### Summary of changes

#### LFP Viewer
- Improve drawing for low-sample-rate sources (<5 kHz)
- Minimize size of `screenBuffer` and `displayBuffer` to lower memory footprint
- Improve performance of triggered display mode

#### Bug fixes
- Fix bug when loading signal chains with multiple splitters. Closes #436 
- Fix record status bug with Rhythm FPGA / Intan Rec. Controller ADC channels. Closes #438
commit 6fcf0496 (2021-04-29): Fix bug when loading signal chains with multiple splitters.

Closes #436.
closed 2021-05-03 by anjaldoshi