open-ephys/plugin-GUI #437: Base station boot firmware version mismatch: 186.173.61453, expected >= 1.0.130

catubc, 2021-04-22:
Hello

We're running into this weird firmware version mismatch on startup.  I saw JSiegle had done the same a few weeks ago, but there is no posted solution.

There is a error in the HST Module Detected! display that says:

REC_NRESET------------------- FAILED w/ error code: 14

And the printout below.

Any advice on what might be going on? The NI PXIe 1071 chassis has firmware version 1.0.0f0, but I couldn't find any other versions.

Thanks so much,
catubc

```

Debug console...

Created processor graph.

Got audio device.
Audio device name: DELL U2520D (NVIDIA High Definition Audio)
Audio device type: Windows Audio
Audio output channels: 2
Audio device sample rate: 44100
Audio device buffer size: 1024
Created audio component.
Created message center.
Created info label.
Created graph viewer.
Adding tab with index 1
Adding tab with index 2
Created data viewport.
Created editor viewport.
Created control panel.
Created filter list.
Created plugin manager
Loading Plugin: ArduinoOutput...
Loading Plugin: BasicSpikeDisplay...
Loading Plugin: CAR...
Loading Plugin: ChannelMappingNode...
Loading Plugin: EvntTrigAvg...
Loading Plugin: FilterNode...
Loading Plugin: IntanRecordingController...
Loading Plugin: LfpDisplayNode...
Loading Plugin: LfpDisplayNodeBeta...
Loading Plugin: PhaseDetector...
Loading Plugin: PulsePalOutput...
Loading Plugin: RecordControl...
Loading Plugin: Rectifier...
Loading Plugin: RhythmNode...
Loading Plugin: SerialInput...
Loading Plugin: SpikeSorter...
Loading Plugin: neuropixels-3a...
Loading Plugin: neuropixels-pxi...
Num built in engines: 2
Adding engine: Binary
Adding engine: Open Ephys
Num plugin engines: 0

Loading window bounds.

Comparing configs
Loading processor graph.
Creating processor with name: Neuropix-PXI
Creating from plugin info...
Neuropix-PXI(1)::Neuropix-PXI
[Neuropixels-PXI] Scanning for devices...
[Neuropixels-PXI]   Found 1 device.
[Neuropixels-PXI]   Opened BS on slot 2
[Neuropixels-PXI]     Searching for probes...
[Neuropixels-PXI]       Found 1.0 single-dock headstage on port: 1
[Neuropixels-PXI] Test module detected
Num engines: 2
Got creator
[Neuropixels-PXI]     Found 0 probes.
[Neuropixels-PXI] Checking for V1 basestations...
[Neuropixels-PXI]   Found V1 Basestation
Base station boot firmware version mismatch: 186.173.61453, expected >= 1.0.130
```

jsiegle, 2021-04-22:
Do you know which version of SpikeGLX you're using? There is a 1:1 correspondence between SpikeGLX version and basestation firmware version, whereas the Open Ephys Neuropixels-PXI plugin works with multiple firmware versions.

What I believe is happening is that it's correctly detecting the HST module, but since the HST tests are not supported with the most recent firmware version, it ignores that basestation and searches for another one. It then finds the same basestation again (this is a bug we need to fix), but tries to communicate with it via the older API, triggering the version mismatch error.

catubc, 2021-04-27:
Hi
Yes, this is the prinouts from start up and help menus:
```

Thd 19312 CPU 6 4/27/21 13:55:56.802] SpikeGLX v.20201103, Imec phase30 v3.31

SpikeGLX v.20201103, Imec phase30 v3.31
Author: Bill Karsh

Based on the SpikeGL extracellular data acquisition system originally developed by Calin A. Culianu.

Copyright (c) 2020, Howard Hughes Medical Institute, All rights reserved.

Use is subject to Janelia Research Campus Software Copyright 1.2 license terms:
http://license.janelia.org/license.

QLed components are subject to GNU Library GPL v2.0 terms:
```
What do you recommend?  We would prefer to use open-ephys with our recordings.

Thanks!

jsiegle, 2021-04-27:
Ok, that's the latest version, so it makes sense that the headstage test module doesn't work.

If you plug in an actual probe, does Open Ephys detect it correctly?

catubc, 2021-04-29:
Yes, indeed, if we plug in a real probe there are no errors.
Thanks for the help!

closed 2021-04-29 by catubc