open-ephys/plugin-GUI #438: "Unwanted" ADC streams get recorded

biomath, 2021-04-23:
Hi,
Intan Recording Controller + v0.5.4 + Windows 10 here.
Even though I'm selecting only one ADC stream in my Record Node, all streams get saved in Open Ephys data format. Channels 1-64 are data channels; Channels 65-67 are non-connected; Channels 68-75 are ADC. Screenshots attached.
Thanks!

[image]

[image]

jsiegle, 2021-04-25:
This bug is fixed in this branch: https://github.com/jsiegle/plugin-GUI/tree/0.5.4-pre-release

@medengineer, can you double check that everything looks ok? I just made sure the ADC channels are added to the `dataChannelOrder` map, because previously they were being ignored.

jsiegle, 2021-04-25:
We will make sure these changes get merged before the next release later this week!

jsiegle, 2021-05-04:
This is fixed in the latest release.

PR #439 (2021-04-29): 0.5.4 pre-release
### Summary of changes

#### LFP Viewer
- Improve drawing for low-sample-rate sources (<5 kHz)
- Minimize size of `screenBuffer` and `displayBuffer` to lower memory footprint
- Improve performance of triggered display mode

#### Bug fixes
- Fix bug when loading signal chains with multiple splitters. Closes #436 
- Fix record status bug with Rhythm FPGA / Intan Rec. Controller ADC channels. Closes #438
closed 2021-05-04 by jsiegle